eclipse-zenoh/zenoh-plugin-dds #10: macOS

joespeed, 2020-10-14:
@kydos @JEnoch heads up that @jwhitleywork and I get same error:
`-- Build files have been written to: /Users/joespeed/Downloads/zenoh-plugin-dds/deps/cyclocut/build
./config.sh: line 40: [: Other: binary operator expected
Scanning dependencies of target cdds-util
[  6%] Building C object CMakeFiles/cdds-util.dir/src/cdds_blob.c.o
In file included from /Users/joespeed/Downloads/zenoh-plugin-dds/deps/cyclocut/src/cdds_blob.c:4:
/Users/joespeed/Downloads/zenoh-plugin-dds/deps/cyclocut/./include/cdds/cdds_util.h:5:10: fatal error: 
      'dds/dds.h' file not found
#include "dds/dds.h"
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/cdds-util.dir/src/cdds_blob.c.o] Error 1
make[1]: *** [CMakeFiles/cdds-util.dir/all] Error 2
make: *** [all] Error 2
Cleaning up.
Joes-MacBook-Pro:zenoh-plugin-dds joespeed$ 
`

JWhitleyWork, 2020-10-14:
The source of this problem is a `bash` syntax error in `config.sh`. #11 fixes this issue.

joespeed, 2020-10-15:
@JEnoch still having an issue macOS 10.15.7

-- Build files have been written to: /Users/joespeed/Downloads/temp3/zenoh-plugin-dds/deps/cyclocut/build
Scanning dependencies of target cdds-util
[  6%] Building C object CMakeFiles/cdds-util.dir/src/cdds_blob.c.o
In file included from /Users/joespeed/Downloads/temp3/zenoh-plugin-dds/deps/cyclocut/src/cdds_blob.c:4:
/Users/joespeed/Downloads/temp3/zenoh-plugin-dds/deps/cyclocut/./include/cdds/cdds_util.h:5:10: fatal error: 
      'dds/dds.h' file not found
#include "dds/dds.h"
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/cdds-util.dir/src/cdds_blob.c.o] Error 1
make[1]: *** [CMakeFiles/cdds-util.dir/all] Error 2
make: *** [all] Error 2
Cargo is already installed, skipping installation.
Cleaning up.

JEnoch, 2020-10-15:
@joespeed : Thus it seems #11 wasn't the solution.
Looking what at config.sh, it's supposed to [build/install CyloneDDS](https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/a488db3cbd078c45839e3f7c2e5e765dc56536e0/config.sh#L16) as a first step.
When you ran ./config.sh, did the build/install of CycloneDDS succeed ? Do you have any file in /usr/local/include/dds ?

JWhitleyWork, 2020-10-15:
@joespeed Can you provide a full log? Something is happening with the installation of Cyclone and the snippet you provided doesn't go back far enough.

joespeed, 2020-10-15:
@JEnoch @JWhitleyWork here is my full log from macOS 10.15.7. You can also see my env at the end 
[zenoh-plugin-dds_config.sh-output.txt](https://github.com/eclipse-zenoh/zenoh-plugin-dds/files/5386373/zenoh-plugin-dds_config.sh-output.txt)

JEnoch, 2020-10-15:
@joespeed looks like CycloneDDS didn't built correctly and thus wasn't installed. See in your log:
```
ld: cannot link directly with dylib/framework, your binary is not an allowed client of /usr/lib/libcrypto.dylib for architecture x86_64
```
The issue looks related to OpenSSL usage. Did you install it via Homebrew ? Otherwise, can you install it with this command:
```bash
brew install openssl@1.1
```

kydos, 2020-12-01:
This builds on MacOS as most of the team runs MacOS. I'll close the issue.

commit a488db3c (2020-10-14): Merge pull request #11 from JWhitleyWork/fix-config.sh. Fix #10

Fix bash Typos and add Cargo Install to config.sh
closed 2020-12-01 by kydos