eclipse-zenoh/zenoh-plugin-dds #13: make zenoh-plugin-dds a ROS or debian package

joespeed, 2020-10-15:
per @JWhitleyWork make Zenoh-plugin-dds a ROS or debian package so it is easily pulled into Autoware.Auto with ROS 2.

JWhitleyWork, 2020-10-15:
Happy to help with the ROS package release process. It can still be built from source outside of ROS but this covers a large number of target ecosystems, I think (`amd64`/`arm64`, Debian, Ubuntu, RedHat, Fedora, MacOS 10.14, Windows 10).

kydos, 2021-02-03:
@esteve maybe you can also help out with this.

esteve, 2021-02-03:
@kydos it'll be a bit tricky to make the plugin a ROS package because it's written in Rust and there's no native way of building a Rust project with Bloom (the tool used for releasing ROS packages). Moreover, the major roadblock would be that the plugin uses nightly features, so the version of the Rust toolchain that Ubuntu ships with wouldn't be enough. One option would be to have a vendor package for Rust nightly and then the plugin package depend on it, but won't be straightforward.

vinnnyr, 2021-08-09:
would hosted container help? https://github.com/eclipse-zenoh/zenoh-plugin-dds/pull/39

JEnoch, 2022-01-21:
I close this issue as both are now supported:
 - [ROS2 package](https://github.com/eclipse-zenoh/zenoh-plugin-dds#ros2-package)
 - [Debian package](https://github.com/eclipse-zenoh/zenoh-plugin-dds/issues/65)

closed 2022-01-21 by JEnoch

esteve, 2022-01-21:
The next step would be submit it to the ROS build farm so that it's available to users, or upload the Debian package to an Apt repository and submit a PR to add it to the ROS index.

JEnoch, 2022-01-21:
I would rather go for the 2nd option, since I already had a try with [ROS build farm](https://index.ros.org/p/zenoh_bridge_dds/github-eclipse-zenoh-zenoh-plugin-dds/#rolling) and found the blooming process really painful. Moreover, as we will anyway release the Debian packages for non-ROS users, I feel pointless to do the same release but with different build system (bloom) for all ROS2 distribs.

Wrt. APT repo, we already have [our own](https://download.eclipse.org/zenoh/debian-repo/) where I'll add the next release of `zenoh-bridge-dds`. 
By _"submit a PR to add it to the ROS index"_ you mean a PR to [rosdep/base.yaml](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml) right ? But I don't see where to add our own repo in some sources list. Do you know if it's possible ?

esteve, 2022-01-21:
> I would rather go for the 2nd option, since I already had a try with ROS build farm and found the blooming process really painful. 

I don't know what your struggles with Bloom were, but it's just a couple of commands. Moreover, if you post the issues you had with Bloom on https://answers.ros.org/questions/, I'm sure someone will be happy to help you.