eclipse-zenoh/zenoh-plugin-dds #16: Topic Whitelisting/Blacklisting

kydos, 2020-11-06:
It would be useful to be able to control by configuration the DDS topics that should not be exposed via zenoh resources or viceversa the set of those that should be.

JEnoch, 2021-09-22:
Whiltelisting is implemented via the [`--allow` option](https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/246e452ed91be7afd9cbd41a19a3dedb2259d664/README.md#limiting-the-ros2-topics-services-parameters-or-actions-to-be-routed).
However blacklisting is not, since the regular expression accepted by this option doesn't support negative matching. The cause being the `regex` crate that we use (see rust-lang/regex#127).

JEnoch, 2022-01-19:
PR #68 adds the `deny` configuration (via config file) and the `--deny` command line option.
If both `allow` and `deny` are set a partition and/or topic will be allowed if it matches only the `allow` expression.

closed 2022-01-19 by JEnoch