eclipse-zenoh/zenoh-plugin-dds #21: Crash when sending compressed video

spiderkeys, 2020-11-28:
After getting the basics working over the internet using std_msgs/msg/String messages, I moved on to trying to send an H264 video stream through the DDS plugin. Discovery and setup within dzd seems to work fine, but the program crashes upon receiving a video frame message.

The frames being sent in this particular test ranged between 20KB - 90KB.

IDL used for video message:
```idl
module test_msgs {
  module msg {
    struct CompressedVideo {
        builtin_interfaces::msg::Time timestamp;    // Timestamp of frame from camera's clock
        unsigned long long            frame_id;     // Monotonically increasing frame counter
        unsigned long                 frame_meta;   // Additional bits for encoding compression-specific metadata
        sequence<octet, 1000000>      data;         // 1MB max allocated size
    };
  };
};
```

Application output:
```
root@a749392:/usr/src/app/zenoh-plugin-dds# RUST_LOG=debug LD_LIBRARY_PATH=/usr/local/lib/ ./target/release/dzd --peer=tcp/172.105.86.91:7447
1606523959.073460 [0]        dzd: using network interface eth0 (udp/10.10.11.3) selected arbitrarily from: eth0
[2020-11-28T00:39:19Z DEBUG zenoh::net] Zenoh Rust API v12bfe92
[2020-11-28T00:39:19Z DEBUG zenoh::net] Config: {"mode": "client", "local_routing": "false", "peer": "tcp/172.105.86.91:7447"}
[2020-11-28T00:39:19Z DEBUG zenoh_router::runtime] Using PID: E4FE6591AE5143689519887B68B72199
[2020-11-28T00:39:19Z DEBUG zenoh_protocol::session::manager] New session opened with 390CF4A4252842FA88EB45C35808C62B: whatami 1, lease 10s, keep alive 1000, sn resolution 268435456, initial sn tx 134184083, initial sn rx 73496218
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::broker] New face 0
[2020-11-28T00:39:19Z DEBUG zenoh_protocol::session::initial] New session link established with 390CF4A4252842FA88EB45C35808C62B: tcp/192.168.1.71:57560 => tcp/172.105.86.91:7447
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::broker] New face 1
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/memory/storage
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/memory/storage/*
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/memory/storage/* for face 0
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/*
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/* for face 0
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/390CF4A4252842FA88EB45C35808C62B/plugin/storages/backend/memory for face 0
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::queries] Register quaryable /@/router/390CF4A4252842FA88EB45C35808C62B for face 0
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /ros_discovery_info
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /ros_discovery_info for face 0
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 4394223598638391801
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 4394223598638391801
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7169910340329578230
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 4394223598638391801
[... 142 lines cut ...]
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /ros_discovery_info for face 1
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Ignoring discovery from local participant: rq/h264_cam_node/describe_parametersRequest
[2020-11-28T00:39:19Z DEBUG dzd] The Subscription(rq/h264_cam_node/describe_parametersRequest, rcl_interfaces::srv::dds_::DescribeParameters_Request_, None is new setting up zenoh and DDS endpoings
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /rq/h264_cam_node/describe_parametersRequest
[2020-11-28T00:39:19Z DEBUG dzd] DiscoveredSubscription(rq/h264_cam_node/get_parameter_typesRequest, rcl_interfaces::srv::dds_::GetParameterTypes_Request_, None
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Ignoring discovery from local participant: rq/h264_cam_node/get_parameter_typesRequest
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /rq/h264_cam_node/describe_parametersRequest for face 1
[2020-11-28T00:39:19Z DEBUG dzd] The Subscription(rq/h264_cam_node/get_parameter_typesRequest, rcl_interfaces::srv::dds_::GetParameterTypes_Request_, None is new setting up zenoh and DDS endpoings
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::resource] Register resource /rq/h264_cam_node/get_parameter_typesRequest
[2020-11-28T00:39:19Z DEBUG dzd] DiscoveredSubscription(rq/h264_cam_node/get_parametersRequest, rcl_interfaces::srv::dds_::GetParameters_Request_, None
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Local Domain Participant IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovery data from Participant with IH = 7069752098226128381
[2020-11-28T00:39:19Z DEBUG zplugin_dds] Discovered endpoint is keyless: true
[... cut ...] sRequest
[2020-11-28T00:39:19Z DEBUG zenoh_router::routing::pubsub] Register subscription /rq/h264_cam_node/get_parameter_typesRequest for face 1
[2020-11-28T00:39:19Z DEBUG dzd] The Subscription(rq/h264_cam_node/get_parametersRequest, rcl_interfaces::srv::dds_::GetParameters_Request_, None is new setting up zenoh and DDS endpoings
dzd: /usr/src/app/zenoh-plugin-dds/deps/cyclocut/src/cdds_blob.c:170: cdds_serdata_from_ser: Assertion `fragchain->nextfrag == NULL' failed.
Aborted (core dumped)
```

kydos, 2020-11-30:
This is failing due to an assertion that ensure that the data received on the DDS side is not fragmented. The workaround is configure DDS to avoid fragmentation -- if possible for your payload. In any support for dealing with fragmented DDS data should come next.

spiderkeys, 2020-11-30:
Thanks @kydos.

Confirmed that after increasing the Cyclone DDS fragment size and reducing the max frame encountered in the video stream to below the fragment size, I was able to make it work. 

Example config:
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <NetworkInterfaceAddress>auto</NetworkInterfaceAddress>
            <AllowMulticast>default</AllowMulticast>
            <MaxMessageSize>65500B</MaxMessageSize>
            <FragmentSize>64000B</FragmentSize>
        </General>
        <Internal>
            <Watermarks>
                <WhcHigh>500kB</WhcHigh>
            </Watermarks>
        </Internal>
        <Tracing>
            <Verbosity>config</Verbosity>
            <OutputFile>stdout</OutputFile>
        </Tracing>
    </Domain>
</CycloneDDS>
```

closed 2020-11-30 by spiderkeys