eclipse-zenoh/zenoh-plugin-dds #24: Cargo build error

maxlein, 2021-02-08:
I am new to cargo, but following the Readme I get following error and I don't really know how to solve this:

```
cargo +nightly build --release --all-targets
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (signal: 11, SIGSEGV: invalid memory reference)
```

JEnoch, 2021-02-09:
Hi @maxlein ,

Could you please provide the following details:

- What is your platform (architecture + OS) ?
- How did you install Rust and Cargo ?
- If you installed `rustup`, can you give us the result of command `rustup show` ?

maxlein, 2021-02-09:
Ubuntu 20.04.1 LTS  
I installed it like described in the readme and ran the script to install cargo.
I am only using the ros2 version of cyclone v0.7.

```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/max/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
nightly-x86_64-unknown-linux-gnu

active toolchain
----------------

stable-x86_64-unknown-linux-gnu (default)
rustc 1.49.0 (e1884a8e3 2020-12-29)

```

JEnoch, 2021-02-09:
This could be a temporary issue with the `rustc` compiler in the nightly toolchain from 2020-12-29.
Please run the `rustup update` command and try to build again.

kydos, 2021-02-22:
@maxlein any updates? Please let us know if this is still an issue, and if not perhaps we can close it.

maxlein, 2021-03-02:
Just tried again with   `nightly-x86_64-unknown-linux-gnu updated - rustc 1.52.0-nightly (4f20caa62 2021-03-01) (from rustc 1.51.0-nightly (04caa632d 2021-01-30))`

Same error:  

```
cargo +nightly build --release --all-targets
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (signal: 11, SIGSEGV: invalid memory reference)
```

kydos, 2021-03-10:
@maxlein that looks like an issue in your rust installation or environment setup. Did you source  ```~/.cargo/env``` ?

To make sure your rust environment is properly setup try the following:

```
$ cargo new demo
$ cargo build
$ cargo run
Hello, world!
```

Let us know what you get.

maxlein, 2021-03-10:
Thanks for your help!  

I explicitly sourced the environment and I get the same error with above commands:  

```
cargo build
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (signal: 11, SIGSEGV: invalid memory reference)

```

akila-ganlath, 2021-03-25:
@kydos I am getting a separate error than what @maxlein experienced:

`error: failed to run custom build command for `cyclors v0.1.0 (https://github.com/kydos/cyclors#89f57025)`

Caused by:
  process didn't exit successfully: `/home/akila/Workspaces/zenoh_ws/zenoh-plugin-dds/target/release/build/cyclors-b1ea1e4cb01db516/build-script-build` (exit code: 101)
  --- stdout
  cargo:rustc-link-lib=ddsc
  cargo:rustc-link-lib=cdds-util

  --- stderr
  warning: argument unused during compilation: '-L/opt/ros/foxy/lib/x86_64-linux-gnu' [-Wunused-command-line-argument]
  /opt/ros/foxy/include/dds/dds.h:15:10: fatal error: 'stddef.h' file not found
  warning: argument unused during compilation: '-L/opt/ros/foxy/lib/x86_64-linux-gnu' [-Wunused-command-line-argument], err: false
  /opt/ros/foxy/include/dds/dds.h:15:10: fatal error: 'stddef.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', /home/akila/.cargo/git/checkouts/cyclors-fbee27736ff49651/89f5702/build.rs:21:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`
  
  ### Environment

- Ubuntu 20.04 x64
- Rust installed [as per](https://www.rust-lang.org/tools/install)
- - cargo installed and tested via cargo hello world demo
- CycloneDDS installed via ROS2 RMW

### Installation

- followed zenoh-plugin-dds README.md

JEnoch, 2021-03-29:
Hi @akila-ganlath,
I think you're missing the `llvm-dev` and `clang-dev` packages that are required by `cyclors` (Rust binding of CycloneDDS). Please let me know if installing those packages solved your issue, and I'll update the README accordingly.

JEnoch, 2021-03-29:
Hi @maxlein,

We didn't manage to reproduce the error you get on 20.04.1 LTS.
As far as I understood you also get it outside of the `zenoh-plugin-dds` project, with the simple Cargo's demo project, right ?

Maybe your issue is similar to https://github.com/rust-lang/rust/issues/80725.
I suggest you look at the backtrace in the dumped core and check if it looks like the one in https://github.com/rust-lang/rust/issues/80725#issuecomment-754763158. If they are similar (i.e. ending in llvm and/or clang) you can try to update or re-install your `llvm-dev` and `clang-dev` packages.

If this doesn't solve your issue, you should post a new issue in https://github.com/rust-lang/rust/issues.

akila-ganlath, 2021-03-30:
> Hi @akila-ganlath,
> I think you're missing the `llvm-dev` and `clang-dev` packages that are required by `cyclors` (Rust binding of CycloneDDS). Please let me know if installing those packages solved your issue, and I'll update the README accordingly.

@JEnoch , installing `clang` and `llvm-dev` via `apt` allowed me to build and run the demo. Thanks for your help!

maxlein, 2021-04-01:
> If this doesn't solve your issue, you should post a new issue in https://github.com/rust-lang/rust/issues.

Ok, I went this way.

closed 2021-04-01 by maxlein