eclipse-zenoh/zenoh-plugin-dds #25: Error after running config.sh: "fatal error: dds/dds.h: No such file or directory"

schornakj, 2021-02-17:
OS: Ubuntu 20.04
ROS: Foxy, installed from binaries
I installed Rust using `rustup` as recommended in this repo's README.

After cloning the repo, I ran the `config.sh` script, but I encounter an error while building some dependencies:

```
In file included from /home/jschornak/zenoh-plugin-dds/deps/cyclocut/src/cdds_blob.c:4:
/home/jschornak/zenoh-plugin-dds/deps/cyclocut/./include/cdds/cdds_util.h:5:10: fatal error: dds/dds.h: No such file or directory
 #include "dds/dds.h"
          ^~~~~~~~~~~
compilation terminated.
```
This ultimately causes the Rust build to fail:

```
error: failed to run custom build command for `cyclors v0.1.0 (https://github.com/kydos/cyclors#e20f3df2)`

Caused by:
  process didn't exit successfully: `/home/jschornak/zenoh-plugin-dds/target/release/build/cyclors-7ce0470c2d0478f1/build-script-build` (exit code: 101)
  --- stdout
  cargo:rustc-link-lib=ddsc
  cargo:rustc-link-lib=cdds-util

  --- stderr
  wrapper.h:1:10: fatal error: 'cdds/cdds_util.h' file not found
  wrapper.h:1:10: fatal error: 'cdds/cdds_util.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', /home/jschornak/.cargo/git/checkouts/cyclors-fbee27736ff49651/e20f3df/build.rs:14:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

kydos, 2021-02-17:
The commit above, in combination with an update to the CMakeLists.txt of the Cyclocut should solve your problem. Let me know and we'll close the issue if that is the case.

schornakj, 2021-02-17:
The commit fixed the error I was seeing when running `config.sh`, but I'm still encountering a build failure:

```
$ cargo +nightly build --release --all-targets
   Compiling cyclors v0.1.0 (https://github.com/kydos/cyclors#e20f3df2)
   Compiling pnet_macros v0.27.2
   Compiling http-types v2.10.0
   Compiling uhlc v0.2.1
error: failed to run custom build command for `cyclors v0.1.0 (https://github.com/kydos/cyclors#e20f3df2)`

Caused by:
  process didn't exit successfully: `/home/jschornak/zenoh-plugin-dds/target/release/build/cyclors-7ce0470c2d0478f1/build-script-build` (exit code: 101)
  --- stdout
  cargo:rustc-link-lib=ddsc
  cargo:rustc-link-lib=cdds-util

  --- stderr
  /usr/local/include/cdds/cdds_util.h:5:10: fatal error: 'dds/dds.h' file not found
  /usr/local/include/cdds/cdds_util.h:5:10: fatal error: 'dds/dds.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', /home/jschornak/.cargo/git/checkouts/cyclors-fbee27736ff49651/e20f3df/build.rs:14:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

kydos, 2021-02-17:
That's coming from the rust crate that wraps cyclocut (cyclone utils). I'll push a fix shortly.

kydos, 2021-02-17:
There is a fix in place now,  let me know if that workout.

schornakj, 2021-02-18:
At risk of peppering you with incremental updates, I now get library linking errors at a later step in the build process. For example:

```
error: linking with `cc` failed: exit code: 1
  |
  = note: "cc" "-Wl,--as-needed" "-Wl,-z,noexecstack" "-m64" "-Wl,--eh-frame-hdr" "-L" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-9eb8bf864a595057.dzd.2gtsdc16-cgu.0.rcgu.o" "-o" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-9eb8bf864a595057" "-Wl,--gc-sections" "-pie" "-Wl,-zrelro" "-Wl,-znow" "-Wl,-O1" "-nodefaultlibs" "-L" "/home/jschornak/zenoh-plugin-dds/target/release/deps" "-L" "/usr/local/lib" "-L" "/home/jschornak/zenoh-plugin-dds/target/release/build/ring-ea13a32ccc75dccc/out" "-L" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/tmp/rustcfX1dzQ/libring-d9865621216192f6.rlib" "-Wl,--start-group" "-Wl,--end-group" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-ea377e9224b11a8a.rlib" "-Wl,-Bdynamic" "-lcdds-util" "-ldl" "-lddsc" "-lcdds-util" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc"
  = note: /usr/bin/ld: cannot find -lddsc
          collect2: error: ld returned 1 exit status
          
error: aborting due to previous error
error: build failed
```

I think it would be good to add this type of environment configuration as an additional job in the CI pipeline, since most ROS2 users will be using package binary distributions.

schornakj, 2021-02-18:
Looks like this issue is related to the feature request in #23.

kydos, 2021-02-18:
@schornakj, that error was due to a small issue on the build.rs in the cyclors library. I've pushed a fix for that and some small updates in the config.sh. It seems to work fine for me on a container cyclone installed through ROS2 distribution (make sure to define ROS_DISTRO if you do not have the full ROS installed). Let me know.

schornakj, 2021-02-18:
I do still get similar build errors using the latest version of the project. I've provided the full logs from my build below.

```
jschornak@jschornak-PickNik-ThinkPad-P15:~/zenoh-plugin-dds$ env | grep CYCLONE
CYCLONE_INCLUDE=/opt/ros/foxy/include
CYCLONE_LIB=/opt/ros/foxy/lib/x86_64-linux-gnu
jschornak@jschornak-PickNik-ThinkPad-P15:~/zenoh-plugin-dds$ env | grep ROS_DISTRO
ROS_DISTRO=foxy
jschornak@jschornak-PickNik-ThinkPad-P15:~$ git clone https://github.com/eclipse-zenoh/zenoh-plugin-dds.git
Cloning into 'zenoh-plugin-dds'...
remote: Enumerating objects: 133, done.
remote: Counting objects: 100% (133/133), done.
remote: Compressing objects: 100% (88/88), done.
remote: Total 293 (delta 60), reused 89 (delta 31), pack-reused 160
Receiving objects: 100% (293/293), 96.01 KiB | 2.09 MiB/s, done.
Resolving deltas: 100% (124/124), done.
jschornak@jschornak-PickNik-ThinkPad-P15:~$ cd zenoh-plugin-dds/
jschornak@jschornak-PickNik-ThinkPad-P15:~/zenoh-plugin-dds$ ./config.sh 
Looking for Cyclone installation at: 
[-.-]
Cyclocut is already installed, skipping installation.
[-.-]
Cargo is already installed, setting up nightly.
info: using existing install for 'nightly-x86_64-unknown-linux-gnu'
info: default toolchain set to 'nightly-x86_64-unknown-linux-gnu'

  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.52.0-nightly (d1206f950 2021-02-15)

info: note that the toolchain 'nightly-2020-12-29-x86_64-unknown-linux-gnu' is currently in use (overridden by '/home/jschornak/zenoh-plugin-dds/rust-toolchain')
Cleaning up.
Done [^_^]

Please set the following environment variables
    export CYCLONE_INCLUDE=/opt/ros/foxy/include
    export CYCLONE_LIB=/opt/ros/foxy/lib/x86_64-linux-gnu

Then run:
    "cargo +nightly build --release --all-targets" 

If you have any questions reach us out on https://gitter.im/atolab/zenoh
jschornak@jschornak-PickNik-ThinkPad-P15:~/zenoh-plugin-dds$ cargo +nightly build --release --all-targets
    Updating crates.io index
    Updating git repository `https://github.com/kydos/cyclors`
    Updating git repository `https://github.com/eclipse-zenoh/zenoh`
  Downloaded idna v0.2.2
  Downloaded async-channel v1.6.1
  Downloaded 2 crates (283.6 KB) in 0.35s
   Compiling libc v0.2.86
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling version_check v0.9.2
   Compiling memchr v2.3.4
   Compiling value-bag v1.0.0-alpha.6
   Compiling log v0.4.14
   Compiling typenum v1.12.0
   Compiling cfg-if v1.0.0
   Compiling serde_derive v1.0.123
   Compiling serde v1.0.123
   Compiling proc-macro-hack v0.5.19
   Compiling autocfg v1.0.1
   Compiling bitflags v1.2.1
   Compiling once_cell v1.5.2
   Compiling regex-syntax v0.6.22
   Compiling getrandom v0.2.2
   Compiling rustc-serialize v0.3.24
   Compiling term v0.4.6
   Compiling unicode-xid v0.0.3
   Compiling signal-hook v0.3.4
   Compiling getrandom v0.1.16
[... 191 lines cut ...]

error: could not compile `zplugin-dds`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: linking with `cc` failed: exit code: 1
  |
  = note: "cc" "-Wl,--as-needed" "-Wl,-z,noexecstack" "-m64" "-Wl,--eh-frame-hdr" "-L" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-c654e54f199682e9.dzd.8yqv1xyy-cgu.0.rcgu.o" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-c654e54f199682e9.dzd.8yqv1xyy-cgu.1.rcgu.o" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-c654e54f199682e9.dzd.8yqv1xyy-cgu.2.rcgu.o" "-o" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-c654e54f199682e9" "/home/jschornak/zenoh-plugin-dds/target/release/deps/dzd-c654e54f199682e9.2n6avoip11d638w3.rcgu.o" "-Wl,--gc-sections" "-pie" "-Wl,-zrelro" "-Wl,-znow" "-Wl,-O1" "-nodefaultlibs" "-L" "/home/jschornak/zenoh-plugin-dds/target/release/deps" "-L" "/usr/local/lib" "-L" "/home/jschornak/zenoh-plugin-dds/target/release/build/ring-ea13a32ccc75dccc/out" "-L" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libenv_logger-c1e1383070216d4a.rlib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libtermcolor-db8b72d8230c5500.rlib" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libtest-f0e2f3090c8b9410.rlib" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libterm-bb4c17ac1c56faac.rlib" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libgetopts-65bd06936d117871.rlib" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libunicode_width-36e93a55c535bf8b.rlib" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_std_workspace_std-b50c9929f7e246da.rlib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libzplugin_dds-7d931ebe6307b50b.rlib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libzenoh-1420eb94259422b3.rlib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libregex-e165a707943c6be9.rlib" "/home/jschornak/zenoh-plugin-dds/target/release/deps/libthread_local-18a02b19280f7005.rlib" "/home/jschornak/z [... cut ...] 9865621216192f6.rlib" "-Wl,--start-group" "-Wl,--end-group" "/home/jschornak/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-ea377e9224b11a8a.rlib" "-Wl,-Bdynamic" "-lcdds-util" "-ldl" "-lddsc" "-lcdds-util" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc"
  = note: /usr/bin/ld: cannot find -lddsc
          collect2: error: ld returned 1 exit status
          

error: aborting due to previous error

error: build failed
```

schornakj, 2021-02-18:
Here is some more info about my specific platform:
```
jschornak@jschornak-PickNik-ThinkPad-P15:~$ uname -a
Linux jschornak-PickNik-ThinkPad-P15 5.8.0-43-generic #49~20.04.1-Ubuntu SMP Fri Feb 5 09:57:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

kydos, 2021-02-18:
Just committed an update to the build.rs to use the CYCLONE_LIB if defined as the library search path. That should solve the issue above, which I guess was the last left to make sure that the DDS plugin builds with a custom cyclone installation, e.g. ROS2.

schornakj, 2021-02-22:
@kydos The latest version works for me now when building from source. Thank you for the fix!

closed 2021-02-22 by schornakj