eclipse-zenoh/zenoh-plugin-dds #26: Using ROS2 introspective tools (ros2 node list, etc.) with Zenoh

schornakj, 2021-02-23:
I have two sets of ROS2 nodes running on different networks connected through a Zenoh router. When I use commands like `ros2 node list` and `ros2 topic list` on one network, they do not receive info about nodes and topics that are on the other network even though the ROS2 nodes are able to establish publisher/subscriber relationships across the Zenoh connection.

It would be very useful to have access to these tools when using Zenoh, since they make it much easier to debug ROS-related node and networking issues.

kydos, 2021-03-10:
Hello @schornakj, we'll look into that. The simplest would be to make a zenoh query to resolve that information as zenoh discovery works differently to precisely avoid flooding the network with discovery data -- which is what DDS does on the other hand. Anyway, let me look into this and I'll let you know my suggestion.

JEnoch, 2021-05-27:
This issue will be solved by #33.

JEnoch, 2021-09-22:
#33 is now implemented. You can test this new feature using the `--fwd-discovery` option. Mode details [here](https://github.com/eclipse-zenoh/zenoh-plugin-dds#full-support-of-ros-graph-and-topic-lists-via-the-forward-discovery-mode)

closed 2021-09-22 by JEnoch