eclipse-zenoh/zenoh-plugin-dds #52: cmake build --release command fails

spanthr, 2021-11-11:
-- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/build
  running: "cmake" "--build" "." "--target" "install" "--config" "Release" "--"
  [ 66%] Building C object CMakeFiles/cdds-util.dir/src/cdds_blob.c.o
  [ 66%] Building C object CMakeFiles/cdds-util.dir/src/builtin.c.o
  [100%] Linking C static library libcdds-util.a
  [100%] Built target cdds-util
  Install the project...
  -- Install configuration: "Release"
  -- Installing: /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/lib/libcdds-util.a
  -- Installing: /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/include/cdds
  -- Installing: /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/include/cdds/cdds_builtin.h
  -- Installing: /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/include/cdds/cdds_util.h
  cargo:root=/home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build
  cargo:rustc-link-search=native=/home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclocut-build/lib
  cargo:rustc-link-lib=static=cdds-util

  --- stderr
  CMake Warning:
    Manually-specified variables were not used by the project:

      CMAKE_ASM_COMPILER
      CMAKE_ASM_FLAGS
      CMAKE_CXX_COMPILER

  CMake Warning:
    Manually-specified variables were not used by the project:

      CMAKE_ASM_COMPILER
      CMAKE_ASM_FLAGS
      CMAKE_INSTALL_LIBDIR
      CYCLONE_INCLUDE
      CYCLONE_LIB

  /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclonedds-build/include/dds/dds.h:15:10: fatal error: 'stddef.h' file not found
  /home/shaurya/zenoh-plugin-dds/zplugin-dds/target/release/build/cyclors-7b461b7c5ccc9415/out/cyclonedds-build/include/dds/dds.h:15:10: fatal error: 'stddef.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', /home/shaurya/.cargo/registry/src/github.com-1ecc6299db9ec823/cyclors-0.1.1/build.rs:59:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed

spanthr, 2021-11-11:
Compiling cyclors v0.1.1
 Building [=======================> ] 364/369: cyclors(build)

JEnoch, 2021-11-12:
Hi @spanthr,

It seems your host misses the `stddef.h` file.
It's usually provided via the **clang development package**. Depending on your OS, you can install it following the instructions in the [README.md](https://github.com/eclipse-zenoh/zenoh-plugin-dds#how-to-build-it) (see 2nd bullet: *"On Linux, make sure the llvm and clang development packages are installed..."*)

JEnoch, 2022-01-21:
I assume you managed to solve this issue somehow.
Feel free to re-open it if that's not the case.

closed 2022-01-21 by JEnoch