eclipse-zenoh/zenoh-plugin-dds #62: zenoh-plugin-dds source from .profile/ .bashrc

sumitpaulde, 2022-01-13:
Is there any way to source the zenoh-plugin-dds? Every time remembering the whole installation path and running the command is kind of annoying and it is cumbersome for the whole ecosystem, if someone has 3 or 4 hosts and remembers each single installation directory, I guess you understood it. Will it be possible to provide some executable file so that we can export it and use ```zenoh-bridge-dds -d 1 ``` only other than ```~/zenoh-plugin-dds/zenoh-plugin-dds/target/release/zenoh-bridge-dds -d 1```

eliasdc, 2022-01-13:
If your using ros2 you can start the plugin with `ros2 run zenoh_bridge_dds zenoh_bridge_dds -d 1`.

sumitpaulde, 2022-01-13:
Sorry for not mentioning about ROS2. Should I build it newly with
 ```colcon build --packages-select zenoh-bridge-dds``` ? 
Because I have already built it with 
```
git clone https://github.com/eclipse-zenoh/zenoh-plugin-dds.git
cd zenoh-plugin-dds
cargo build --release -p zenoh-bridge-dds
```
If I have to built it newly with the colcon build should I build it from the downloaded directory?

eliasdc, 2022-01-13:
a typical workspace build for zenoh-plugin-dds is the following:
```
cd ~/ros_ws/src/
git clone https://github.com/eclipse-zenoh/zenoh-plugin-dds.git
cd ~/ros_ws
colcon build -DCMAKE_BUILD_TYPE=Release
source setup.bash
```
So colcon builds are done in the workspace and not in the src folder or the zenoh-plugin-dds folder.

sumitpaulde, 2022-01-13:
Hi, @eliasdc Thanks a lot for your kind and quick response.

Running ```colcon build -DCMAKE_BUILD_TYPE=Release``` in the ros2 directory giving me this following error:
```
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,test,test-result,version-check}
              ...
colcon: error: unrecognized arguments: -DCMAKE_BUILD_TYPE=Release
```

eliasdc, 2022-01-13:
My appologies, copy paste error. The correct command is: `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`. Updated the original reply too

sumitpaulde, 2022-01-13:
Hi @eliasdc, Thanks a lot, it is working fine. As I have built the zenoh-plugin-dds  outside of ROS2 and it seems I do not need it anymore, I would like to know how can I remove it, ``` sudo -rm zenoh-plugin-dds ``` will be enough? I would also like to request you to update the https://github.com/eclipse-zenoh/zenoh-plugin-dds#ros2-package so new users like me will be benefitted.

eliasdc, 2022-01-14:
I think you can just remove the `target/release` folder inside your zenoh-plugin-dds folder to remove the cargo build. For removing the ROS2 build you would remove the `build/zenoh-plugin-dds` and `install/zenoh-plugin-dds` folders inside your ros workspace folder. 

The build instructions for ros2 in https://github.com/eclipse-zenoh/zenoh-plugin-dds#ros2-package are correct but it doesn't to an optimized release build.

sumitpaulde, 2022-01-14:
Thanks a lot, @eliasdc for replying patiently and @JEnoch for taking this issue seriously. I am going to close this issue.

closed 2022-01-14 by sumitpaulde