eclipse-zenoh/zenoh-plugin-dds #63: waiting for action server or service broken

eliasdc, 2022-01-13:
When using action servers or services it is typical to first test if the action server or service is available with `wait_for_action_server` or `wait_for_service`. When having a zenoh bridge node active it will create an action server or service proxy which will then trigger the wait to finish and succeed. Navigation stack uses this feature in the life-cycle nodes to wait for action servers or services to become active before activating itself.

code example:
```
rclcpp_action::Client<NavigateToPose>::SharedPtr client =
    rclcpp_action::create_client<NavigateToPose>(this, "navigate_to_pose");
  while (!client->wait_for_action_server(std::chrono::duration<double>(1.0))) {
    RCLCPP_ERROR(
      get_logger(), "timed out waiting for action server");
    if (!rclcpp::ok())
      throw std::runtime_error("exited rclcpp while constructing client");
  }
```

My current work around is to only allow topics to be routed through zenoh using `--allow /rt/'.

JEnoch, 2022-01-14:
I'm not sure what `wait_for_action_server` and `wait_for_service` are waiting for, but I guess it's related to discovery of DDS Readers/Writers implementing the action/service and/or some ROS graph information.

Indeed, this will not work with the default mode of `zenoh-bridge-dds` that doesn't forward any discovery information to the remote bridge.
But did you try using the [forward discovery mode](https://github.com/eclipse-zenoh/zenoh-plugin-dds#full-support-of-ros-graph-and-topic-lists-via-the-forward-discovery-mode) (`-f` option) ?

eliasdc, 2022-01-14:
`wait_for_action_server` and `wait_for_service` would normally wait indefinitely (when no timeout set) until a action server or service is launched. As `zenoh-bridge-dds` makes a proxy for the discovered actions and services it will let the clients know there is a service or action server available even though there is not a real one available yet (local or remote).

The problem is not only related to remote action servers or services but for local ones which haven't been launched yet. Using the forward discovery won't make any difference.

JEnoch, 2022-01-17:
OK, but both `wait_...` operations are specific to the name of the action or service they're waiting for, right ?
So the problem you refer occurs only if actions (or services) with the same name are eventually existing on both sides (remote and local). But the the remote one is started before the local one, making the client to detect that it's launched. Still, you would like the client to specifically wait for the local one, not the remote.

In my view, the root of the problem is the existence of 2 distinct services (or actions) with the same name in the system, don't you think ?
So indeed denying the routing of remote services or actions is a solution to be sure that your client only waits (and calls) the local ones. But do you have another solution in mind ?

eliasdc, 2022-01-17:
Yes, both `wait_...` operations are specific to the name of the action or service. 

The problem exists even if no service or action server is launched. So just a single client using the wait functions. An example on a single host
* start a node which creates a service client and waits for the service to become available --> node will wait forever
* start zenoh-bridge-dds on the same domain --> waiting service will get triggered without there being a real service server in the network
* service client will have found a server, but all commands to it will timeout as there is no real service server available only the 'proxy server' created by zenoh-bridge-dds

JEnoch, 2022-01-17:
I just made a test with the [Actions tutorial](https://docs.ros.org/en/galactic/Tutorials/Actions/Writing-a-Py-Action-Server-Client.html), and I confirm using the `-f` option solves this issue:
 * `python3 fibonacci_action_client.py` (a modified version with logs around the call to `self._action_client.wait_for_server() ` => the client is blocked on the wait
 * `RUST_LOGS=debug zenoh-bridge-dds -f` => the client is still blocked
 * `python3 fibonacci_action_server.py` => the client is unblocked and the server starts the Fibonacci sequence

In default mode, there is indeed an issue since the bridge create local matching Readers/Writers to the action client, faking an action server that doesn't exist.
But in [forward discovery mode](https://github.com/eclipse-zenoh/zenoh-plugin-dds#full-support-of-ros-graph-and-topic-lists-via-the-forward-discovery-mode) the bridge make the remote bridges to create "copies" of client's Readers/Writers faking the action client (not the server). Thus nothing is created locally and the client is still blocked in the wait. It will be unblocked only with a local action server, or a remote one that will make the remote bridge to request the local bridge the creation of "copies" of server's Readers/Writers faking the action server.

eliasdc, 2022-01-17:
Great news you could recreate the issue.

I didn't know forward discovery mode changes the readers and writers creation strategy.

Is the issue in the default mode solvable ? As I want to keep the transferred data to a minimum and that's why I have forward discovery disabled. I will definitely test if it does solve all my issues with navigation stack failing to come up, etc.

JEnoch, 2022-01-17:
In default mode I see 2 solutions to solve this:

1. Assuming you don't want your action/service client to call a remote action/service server, but only a local one: 
   you could just forbid the routing of this action/service. And that's actually what you did in a way via `--allow /rt/`.
   However I realise it would be more user friendly to also have a `--deny` option to allow all but some actions/services. #16 is still open for this, and I'll try to find some time to implement this soon.
2. If you don't want to forbid remote calls to the action/service, but still to have your client unblocked only when a remote action/service is discovered (or a local one without the bridge being involved):
   there is no choice but to transmit this discovery information from the remote bridge to the local bridge. And that's what the  forward discovery mode does.

To reassure you wrt. the amount of discovery data forwarded in this mode: it's not a forwarding of the full DDSI's SPDP/SEDP protocol. The bridge just sends over zenoh 1 message per discovered/undiscovered Reader/Writer (with its QoS in a compact way), not X messages for the X participants on the other side as it would have been the case with SEDP.

The `ros_discovery_info` are also forwarded but in a consolidated and less intensive way: the bridge reads on this topic every 500ms and forwards only the latest publication for each participant gid.
This drastically reduces the traffic on this topic when a participant starts with several ROS2 nodes and topics/services/actions (most of the cases AFAIK).

Let me know the results of your tests with forward discovery mode, for both functional and traffic aspects.

JEnoch, 2022-01-19:
I just added the `--deny` option via #68.

How are going your tests with forward discovery mode ?

eliasdc, 2022-03-11:
Thanks for the `--deny` option. I've been using that to workaround my problems.

I tested the forward discovery mode and it does indeed solve the problem of the action clients. But for me this isn't usable as in that mode `TRANSIENT_LOCAL` topics are not working at all.

JEnoch, 2022-03-16:
Thanks for your update.

wrt. `TRANSIENT_LOCAL` topics, do you refer to issue #61 ?
If not, could you please create a new issue with details of your use case, commands and misbehaviour you see ?

closed 2022-03-16 by JEnoch

eliasdc, 2022-03-16:
yes I think it is related to #61.