eclipse-zenoh/zenoh-plugin-dds #73: How to choose network interface where plugin runs?

ipa-rwu, 2022-02-03:
From the log as shown below, I see the plugin chooses "wlan0". But I want to run on another interface. Can I config it?

```
async-std/: using network interface wlan0 (udp/192.168.0.213) selected arbitrarily from: wlan0, docker0, flannel.1, cni0
```

Thanks in advance!

JEnoch, 2022-02-04:
Hi @ipa-rwu,

This log comes from CycloneDDS that the plugin uses internally for **DDS communications**.
You can configure it in the same way than any application using CycloneDDS, i.e. via the [CYCLONEDDS_URI](https://github.com/eclipse-cyclonedds/cyclonedds#configuration) environment variable.

Thus, to set the network interface used for DDS you can add the [//CycloneDDS/Domain/General/NetworkInterfaceAddress](https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/docs/manual/options.md#cycloneddsdomaingeneralnetworkinterfaceaddress) to the XML configuration of CycloneDDS.
For instance (setting directly the XML content in the CYCLONEDDS_URI environment variable) you can run `zenoh-bridge-dds` as such:
```
CYCLONEDDS_URI='<CycloneDDS><Domain><General><NetworkInterfaceAddress>cni0</NetworkInterfaceAddress></General></Domain></CycloneDDS>' zenoh-bridge-dds
```

For the **zenoh communications**, by default `zenoh-bridge-dds` opens a TCP server socket on ANY interface (0.0.0.0) and a random port. You can change this via the `-l` option. For instance:
```
zenoh-bridge-dds -l tcp/127.0.0.1:4567
```
For scouting protocol (i.e. discovery of other zenoh peers and routers) it joins the UDP multicast group `224.0.0.224:7447` on all interfaces. You can deactivate the scouting using the `--no-multicast-scouting` option.

Finally, activating the debug logs (`RUST_LOG=debug` environment variable) you will see all the used interfaces and ports in the very first logs.

ipa-rwu, 2022-02-09:
@JEnoch Thanks a lot! It works for me.

closed 2022-02-09 by ipa-rwu