eclipse-zenoh/zenoh-plugin-dds #100: [Bug] rustc version installed through rosdep is lagging behind that required by this package

JohnTGZ, 2022-11-08:
### Describe the bug

`rustc` version installed through `rosdep` is version `1.61.0` but package `zenoh-config v0.6.0-beta.1` requires `rustc 1.62.1 or newer`.

If rust is installed for the first time via rosdep, `rustup` will not be installed and cannot be used to update the rustc version. The workaround now is to install rustup and rustc packages the [officially recommended way](https://www.rust-lang.org/tools/install):
```bash
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
```

### To reproduce

1. Create workspace and clone the repo 
```bash
mkdir -p zenoh_ws/src
git clone https://github.com/eclipse-zenoh/zenoh-plugin-dds.git zenoh_ws/src/zenoh-plugin-dds
```
2. Download and install dependencies through rosdep. This installs the packages `cargo` and `rustc`
```bash
cd zenoh_ws
rosdep install --from-paths src --ignore-src -r
```
3. Build the workspace
```bash
colcon build
```
4. Error shows
```bash
error: package `zenoh-config v0.6.0-beta.1` cannot be built because it requires rustc 1.62.1 or newer, while the currently active rustc version is 1.61.0
gmake[2]: *** [CMakeFiles/build_crate.dir/build.make:79: target/release/zenoh-bridge-dds] Error 101
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/build_crate.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zenoh_bridge_dds [1min 22s, exited with code 2]

Summary: 0 packages finished [1min 22s]
  1 package failed: zenoh_bridge_dds
  1 package had stderr output: zenoh_bridge_dds

```

### System info

- Platform: [Ubuntu 22.04.1 LTS]
- CPU: [AMD Ryzen 7 3800x]
- Zenoh version: 0.6.0-beta.1

JEnoch, 2022-11-17:
The Rust eco-system is moving fast with regular releases of the Rust toolchains. Each release brings improvements in terms of features, stabilisations, safety improvements and performances.
In Zenoh project, we would like to benefit those improvements, not being stuck with support of *"old"* Rust versions. 
And I guess the same might go for some dependencies used by Zenoh (or other ROS/Rust packages) that at any point could decide to require a recent Rust version.

As you wrote, the recommended way to install the Rust environment is via `rustup`. 
I think a long term solution to have `rustup` installed via `rosdep`, and Rust projects depending on it, rather than on the OS `rust` package that could be *"outdated"* (e.g. `1.48.0` on Debian bullseye).

JohnTGZ, 2022-11-18:
I agree with you, that makes complete sense. I raised this issue in the event that other users doing `rosdep install` will be aware of this, especially if they are using it in their CI workflow.

closed 2023-11-28 by JEnoch

JohnTGZ, 2023-11-30:
Got it. Thanks for the clarification and support.