eclipse-zenoh/zenoh-plugin-dds #101: [Bug] Unable to build with the latest zenoh-plugin-dds

evshary, 2022-11-09:
### Describe the bug

While building the latest version of zenoh-plugin-dds, it causes the following error:

```
.....
error: failed to run custom build command for `cyclors v0.1.5`                                                                                                         
                                                                                                                                                                       
Caused by:                                                                                                                                                             
  process didn't exit successfully: `/home/ros/workspace/zenoh-plugin-dds/target/release/build/cyclors-073c38f7381a1401/build-script-build` (exit status: 101)         
  --- stdout                                                                                                                                                           
  CMAKE_TOOLCHAIN_FILE_x86_64-unknown-linux-gnu = None                                                                                                                 
  CMAKE_TOOLCHAIN_FILE_x86_64_unknown_linux_gnu = None                                                                                                                 
  HOST_CMAKE_TOOLCHAIN_FILE = None                                                                                                                                     
  CMAKE_TOOLCHAIN_FILE = None                                                                                                                                          
  CMAKE_GENERATOR_x86_64-unknown-linux-gnu = None                                                                                                                      
  CMAKE_GENERATOR_x86_64_unknown_linux_gnu = None                                  
  HOST_CMAKE_GENERATOR = None                                                      
  CMAKE_GENERATOR = None                                                                                                                                               
  CMAKE_PREFIX_PATH_x86_64-unknown-linux-gnu = None                                
  CMAKE_PREFIX_PATH_x86_64_unknown_linux_gnu = None                                                                                                                    
  HOST_CMAKE_PREFIX_PATH = None                                                                                                                                        
  CMAKE_PREFIX_PATH = None                                                         
  CMAKE_x86_64-unknown-linux-gnu = None                                                                                                                                
  CMAKE_x86_64_unknown_linux_gnu = None                                                                                                                                
  HOST_CMAKE = None                                                                                                                                                    
  CMAKE = None                                                                                                                                                         
  running: "cmake" "/home/ros/.cargo/registry/src/github.com-1ecc6299db9ec823/cyclors-0.1.5/cyclonedds" "-DBUILD_SHARED_LIBS=OFF" "-DBUILD_IDLC=OFF" "-DBUILD_DDSPERF=O
FF" "-DENABLE_SSL=NO" "-DENABLE_SECURITY=NO" "-DCMAKE_INSTALL_LIBDIR=lib" "-DCMAKE_INSTALL_PREFIX=/home/ros/workspace/zenoh-plugin-dds/target/release/build/cyclors-662
b0f66ac098368/out/cyclonedds-build" "-DCMAKE_C_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_C_COMPILER=/usr/bin/cc" "-DCMAKE_CXX_FLAGS= -ffunction-s
ections -fdata-sections -fPIC -m64" "-DCMAKE_CXX_COMPILER=/usr/bin/c++" "-DCMAKE_ASM_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_ASM_COMPILER=/usr/
bin/cc" "-DCMAKE_BUILD_TYPE=Release"                                                                                                                                   
  -- The C compiler identification is GNU 9.4.0                                                                                                                        
  -- Check for working C compiler: /usr/bin/cc                                                                                                                         
  -- Check for working C compiler: /usr/bin/cc -- works                                                                                                                
  -- Detecting C compiler ABI info                                                 
  -- Detecting C compiler ABI info - done                                                                                                                              
  -- Detecting C compile features                                                                                                                                      
  -- Detecting C compile features - done                                                                                                                               
  -- Looking for getopt.h                                                          
  -- Looking for getopt.h - found                                                                                                                                      
  -- Building without OMG DDS Security support
......
```

I guess the error is from the commit [Use new cyclors](https://github.com/eclipse-zenoh/zenoh-plugin-dds/commit/6ae7447de589ebaaed0210f30343760df28101a2).
If I use the previous commit, the error disappears.

### To reproduce

1. Get the code from GitHub
2. Build standalone executable
```bash
cargo build --release -p zenoh-bridge-dds
```

### System info

- Platform: Ubuntu 20.04
- CPU: Intel(R) Core(TM) i5-8365U CPU @ 1.60GHz

evshary, 2022-11-09:
Thanks for @JEnoch 's solution. It seems like I can't source ROS environment first. ROS environment includes CycloneDDS and iceoryx. If I built zenoh-bridge-dds with this, it'll try to build with iceoryx and will fail.
However, I still fail to build zenoh-bridge-dds with the following error:

```
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "/tmp/rustcsKR4eT/symbols.o" "/home/ros/zenoh-plugin-dds/target/release/deps/zenoh_bridge_dds-81ed62f3e2c6e910.zenoh_bridge_dds.ff90c0dd-cgu.0.rcgu.o" "-Wl,--as-needed" "-L" "/home/ros/zenoh-plugin-dds/target/release/deps" "-L" "/home/ros/zenoh-plugin-dds/target/release/build/ring-b7127fa6e8b7f748/out" "-L" "/home/ros/zenoh-plugin-dds/target/release/build/blake3-60e3804d50dd7f29/out" "-L" "/home/ros/zenoh-plugin-dds/target/release/build/blake3-60e3804d50dd7f29/out" "-L" "/home/ros/zenoh-plugin-dds/target/release/build/cyclors-662b0f66ac098368/out/cyclonedds-build/lib" "-L" "/home/ros/zenoh-plugin-dds/target/release/build/cyclors-662b0f66ac098368/out/cyclocut-build/lib" "-L" "/home/ros/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/tmp/rustcsKR4eT/libblake3-50e98199e1b8d584.rlib" "/tmp/rustcsKR4eT/libcyclors-a3fe6eb29282c4bd.rlib" "/tmp/rustcsKR4eT/libring-944069f5d5422c53.rlib" "/home/ros/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-272ca28f0b8538d5.rlib" "-Wl,-Bdynamic" "-ldl" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-znoexecstack" "-L" "/home/ros/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/ros/zenoh-plugin-dds/target/release/deps/zenoh_bridge_dds-81ed62f3e2c6e910" "-Wl,--gc-sections" "-pie" "-Wl,-zrelro,-znow" "-Wl,-O1" "-nodefaultlibs"
  = note: /usr/bin/ld: /home/ros/zenoh-plugin-dds/target/release/deps/zenoh_bridge_dds-81ed62f3e2c6e910.zenoh_bridge_dds.ff90c0dd-cgu.0.rcgu.o: in function `zplugin_dds::dds_mgt::on_data':
          zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0x27): undefined reference to `dds_get_participant'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0x43): undefined reference to `dds_get_instance_handle'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0xf3): undefined reference to `dds_take'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0x7d8): undefined reference to `dds_qget_reliability'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0x984): undefined reference to `dds_qget_reliability'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds7dds_mgt7on_data17h090b60bcae13ba7aE+0xf4f): undefined reference to `dds_return_loan'
          /usr/bin/ld: /home/ros/zenoh-plugin-dds/target/release/deps/zenoh_bridge_dds-81ed62f3e2c6e910.zenoh_bridge_dds.ff90c0dd-cgu.0.rcgu.o: in function `zplugin_dds::qos::Qos::from_qos_native_with_reliability':
          zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x38): undefined reference to `dds_qget_durability'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x128): undefined reference to `dds_qget_durability_service'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x203): undefined reference to `dds_qget_deadline'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x234): undefined reference to `dds_qget_latency_budget'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x267): undefined reference to `dds_qget_ownership'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x2d7): undefined reference to `dds_qget_liveliness'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x333): undefined reference to `dds_qget_destination_order'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x386): undefined reference to `dds_qget_history'
          /usr/bin/ld: zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos32from_qos_native_with_reliability17h292227655b83d169E+0x40a): undefined reference to `dds_qget_partition'
          /usr/bin/ld: /home/ros/zenoh-plugin-dds/target/release/deps/zenoh_bridge_dds-81ed62f3e2c6e910.zenoh_bridge_dds.ff90c0dd-cgu.0.rcgu.o: in function `zplugin_dds::qos::Qos::to_qos_native':
          zenoh_bridge_dds.ff90c0dd-cgu.0:(.text._ZN11zplugin_dds3qos3Qos13to_qos_native17h150d60e8dc07507fE+0x16): undefined reference to `dds_qos_create'
[... cut ...]
          collect2: error: ld returned 1 exit status
          
  = help: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `zenoh-bridge-dds` due to previous error
```

JEnoch, 2022-11-14:
Analysing the differences of build flags between CycloneDDS 0.7.0 and 0.10.2 I found the cause are the new `-flto -fno-fat-lto-objects` flags introduced in CycloneDDS 0.10.2 build.
Those can be deactivated configuring CycloneDDS CMake with `ENABLE_LTO=no`. And doing this, the build succeeds.

This requires flag has to be set in [cyclors](https://github.com/kydos/cyclors) build and a new version of it shall be released.

evshary, 2022-11-15:
Thank you @JEnoch 
I can now build the latest zenoh-bridge-dds successfully!
I'll close the issue.

JEnoch, 2022-11-15:
Fixed by 11212d6b5e20947d643ab5547b0c1ef0c09c5323