eclipse-zenoh/zenoh-plugin-dds #105: [Bug] Zenoh router doesn't seem to work with `@appendable` data

alknemeyer, 2022-11-28:
### Describe the bug

I use Zenoh plugin to access DDS data via VPN (ie the router configuration). This works well when the data types are annotated with `@final` (the current default) but fails when the types are annotated with `@appendable`.

I don't see reference to XTypes in this repository, although I'm not familiar with it (`@appendable` is a relatively new feature).

Is this expected behavior?

### To reproduce

1. Create a data type in an IDL file and annotate it with either `@final` or `@appendable`
2. Create a publisher and a subscriber
3. Run the publisher on computer A, as well as a Zenoh bridge: `docker run --init --rm --net=host eclipse/zenoh-bridge-dds:0.6.0-beta.1 --no-multicast-scouting -l tcp/0.0.0.0:7447`
4. Run a VPN, and run the subscriber on computer B. Run a second Zenoh bridge: `docker run --init --rm --net=host eclipse/zenoh-bridge-dds:0.6.0-beta.1 --no-multicast -scouting -m peer -e tcp/$IP:7447`

If the data type is `@final`, this works. If it's `@appendable`, it doesn't.

### System info

- DDS implementation: CycloneDDS v0.10.2
- VPN: WireGuard
- Platform: CentOS Stream 8
- CPU: AMD Ryzen 9 5950X

JEnoch, 2022-11-29:
The **0.6.0-beta.1** version relies on **CycloneDDS 0.7.0** which wasn't supporting XTypes yet.
Thus CycloneDDS in the Zenoh plugin probably drops the DATA message with XCDR encoding flag, as it's considered as not compatible.
Note that a `@final` type is equivalent to a "non-extended" IDL type, and thus is sent with the CDR flag that is supported by **0.7.0**.

In master branch, `zenoh-plugin-dds` now relies on **CycloneDDS 0.10.2** which supports XTypes.
Can you please test with it using the `eclipse/zenoh-bridge-dds:master` image ?

alknemeyer, 2022-11-29:
I don't know if it's relevant, but this error log is new to me:
```
[2022-11-29T16:23:05Z WARN  zenoh::session] Received ReplyData for unkown Query: 68
[2022-11-29T16:23:05Z WARN  zenoh::session] Received ReplyFinal for unkown Query: 68
[2022-11-29T16:23:05Z WARN  zenoh::session] Received ReplyData for unkown Query: 71
```

stevemcnamaraultra, 2024-07-11:
Would it be possible to confirm that this is still a known issue please?

I am currently suffering this problem - any data marked as @appendable fails to route through the bridge.
An error message is reported by the bridge when it attempts to handle a subscriber created by my application:

2024-07-11T10:15:11.324110Z DEBUG async-std/runtime ThreadId(11) zenoh_plugin_dds::route_zenoh_dds: Route Zenoh->DDS (SERVICEINFO -> SERVICEINFO): create DDS Writer
2024-07-11T10:15:11.324145Z ERROR async-std/runtime ThreadId(11) zenoh_plugin_dds: Route Zenoh->DDS (SERVICEINFO -> SERVICEINFO): creation failed: Error creating DDS Writer: Bad Parameter

If the topic is changed to marked it as @final it works as expected.

I am using the latest CycloneDDS build (that I build myself using type discovery enabled and topic discovery enabled) and a pre-built zenoh_dds_bridge.

Thanks for your assistance with this.

gmartin82, 2025-05-16:
This is now fixed on the latest main: #83225be