eclipse-zenoh/zenoh-plugin-dds #116: Easier to configure the limitation of ROS/DDS topic

evshary, 2023-02-17:
### Describe the feature

When I try to integrate zenoh-bridge-dds into Autoware, I often want to constrain which topic can pass through zenoh-bridge-dds.
In my architecture, I want zenoh-bridge-dds acts like a gateway that transmits necessary ROS topic to the fleet management system, and the rest of the private ROS topic would not be seen by others.

The architecture will be something like the following:
[image]

However, I found that the current ROS/DDS topic limitation is a little unfriendly.
In the description [here](https://github.com/eclipse-zenoh/zenoh-plugin-dds#limiting-the-ros2-topics-services-parameters-or-actions-to-be-routed), we need to use the regular expression to specify the allowed topic.
But while I have too many allowed topics and put all of them in one line, it's hard to maintain in [the configuration file](https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/master/DEFAULT_CONFIG.json5#L47).

IMHO, since we are using json in the configuration file, maybe it's suitable to use list instead of string to list all the allow / deny topic.

```json5
allow: [
  '/rosout',
  '/turtle1/cmd_vel',
  '/turtle1/rotate_absolute',
  ...
],
```

Maybe we can concatenate these topics with '|' internally, and no need to change the code too much.

JEnoch, 2023-02-17:
Thanks for this suggestion! I think that's a good idea.
I'll try to do a PR for this in the coming weeks. Or maybe you would like to contribute ? 😉

evshary, 2023-02-21:
> Thanks for this suggestion! I think that's a good idea. I'll try to do a PR for this in the coming weeks. Or maybe you would like to contribute ? wink

I'm fine with that. Maybe I can take a look at it these days. 😆

JEnoch, 2023-03-06:
Implemented via #119.
Thanks for your contribution @evshary !

closed 2023-03-06 by JEnoch