eclipse-zenoh/zenoh-plugin-dds #117: [Bug] zenoh-plugin-dds build error on Ubuntu 20.04.5 LTS

binbowang1987, 2023-02-22:
### Describe the bug

```
error: linking with `cc` failed: exit status: 1
|                                                                                                                                                                                 
= note: "cc" "-m64" "/tmp/rustcqGsRx3/symbols.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.10lub30xrxmx4633.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1156rw6x2a8zsxr3.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.125lds21uwo7fry7.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.12v8hxijimyc9i5w.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.13ugvcbn2ka91auo.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.171k23vazu4ch0zs.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.17wupo9tj7edmm38.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1a7g6q6chjj4bowx.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1amumrndjpkryrz7.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1anesqodk5zugnkc.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1ay4oawih64xqw33.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1cib57wsqw4fswbp.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1eyuzrr4avai2d5l.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1gbdurnpzx564e97.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1hf8994g3ik973tz.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1k2o48bi79l3hi9t.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1kcjbq2yfpbm040i.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1m1ia4vmnlzwej64.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1mkfx1yjpk7imgo0.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1nfjrpk6pvpga9yh.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1oohvob3o4f8g8ii.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1pcuduotg30y31dx.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1pziqy8uo6xzoa2m.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1q6zu41gdfhafb0e.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1rexd7jqrpr6usfd.rcgu.o" "/home/dev/work/01-dds/rustd
ds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1s1hxkn21adnzttl.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dd
s-cca60c3700773892.1s6ti0gd9589z6co.rcgu.o" "/home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/zenoh_bridge_dds-cca60c3700773892.1umuy0vxftmc2tiz.rcgu.o
........
= note: /usr/bin/ld: /home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/libzplugin_dds.rlib(zplugin_dds.iimurihg1yney8n.rcgu.o): in function `load_plugin':
          /home/dev/.cargo/registry/src/github.com-1ecc6299db9ec823/zenoh-plugin-trait-0.7.0-rc/src/vtable.rs:95: multiple definition of `load_plugin'; /home/dev/work/01-dds/rustdds/zenoh-plugin-dds/target/debug/deps/libzplugin_rest-317e5b3441c671df.rlib(zplugin_rest-317e5b3441c671df.zplugin_rest.0d6cac0f-cgu.8.rcgu.o):/home/dev/.cargo/registry/src/github.com-1ecc6299db9ec823/zenoh-plugin-trait-0.7.0-rc/src/vtable.rs:95: first defined here
          collect2: error: ld returned 1 exit status

error: could not compile `zenoh-bridge-dds` due to previous error

```

### To reproduce

1. git clone
2. cargo build
3. build error

### System info

- Palatform: Ubuntu 20.04.5 LTS 
- CPU: Intel(R) Xeon(R) Silver 4316 CPU @ 2.30GHz
- Zenoh: 0.7.0-rc

JEnoch, 2023-02-22:
Hi @binbowang1987 ,

**TL;DR:**
This error occurs on Linux when building both plugin (`zplugin-dds `) and standalone executable (`zenoh-bridge-dds `) at the same time with the `cargo build` command.
You must build each independently as per [README instructions](https://github.com/eclipse-zenoh/zenoh-plugin-dds#how-to-build-it):
  - `cargo build --release -p zplugin-dds`
  - `cargo build --release -p zenoh-bridge-dds`

I will improve the README to emphasis this.

**Root cause of the issue**:
To be dynamically loaded by the Zenoh router (`zenohd`), a Zenoh plugin must defines a `load_plugin()` operation in C representation and non-mangled.
In the `zenoh-bridge-dds` we rather statically link 2 plugins (`zplugin-dds` and `zplugin-rest`). If those plugins both define the same non-mangled `load_plugin()` operation,`ld` reports a conflict because of multiple definitions. 
Therefore, in all plugins `Cargo.toml`, we define a `no_mangle` feature (active by default) that can can be deactivated when statically linking several plugins in a same bin. `zenoh-bridge-dds/Cargo.toml` well [deactivates this feature](https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/f94ae615874225f4896f3350c78dab007d12c21e/zenoh-bridge-dds/Cargo.toml#L36). But unfortunately, when building both `zplugin-dds` and `zenoh-bridge-dds` together, Cargo build `zplugin-dds` only once and with this `no_mangle` feature enabled.

closed 2023-02-22 by JEnoch

milyin, 2024-01-09:
> **TL;DR:** This error occurs on Linux when building both plugin (`zplugin-dds `) and standalone executable (`zenoh-bridge-dds `) at the same time with the `cargo build` command. You must build each [...]

This issue is fixed in PRs https://github.com/eclipse-zenoh/zenoh-plugin-dds/pull/168, https://github.com/eclipse-zenoh/zenoh/pull/583.

Since these PRs the macro call `zenoh_plugin_trait::declare_plugin!` (https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/master/zenoh-plugin-dds/src/lib.rs#L106) now defines plugin's dynamic library functions only if package's local feature named "no_mangle" is enabled: https://github.com/eclipse-zenoh/zenoh/pull/583/files#diff-7503eb1ae5fdc0a28d3b10a946641fb3f5bd7501be4d8af6e62393b135145d14R206. So `zenoh-bridge-dds` doesn't define these functions anymore and so no linking error occures.

commit b86edeb7 (2023-02-22): README: clarify build instructions (fix #117)