eclipse-zenoh/zenoh-plugin-dds #127: [Bug] Zenoh DDS reporting error when using CycloneDDS localhost config

sisaha9, 2023-04-19:
### Describe the bug

Zenoh peers are unable to communicate with each other. Both machines run ROS 2 Humble and have the below config

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <Interfaces>
                <NetworkInterface name="lo" multicast="true" />
            </Interfaces>
            <AllowMulticast>default</AllowMulticast>
            <MaxMessageSize>65500B</MaxMessageSize>
        </General>
        <Internal>
            <Watermarks>
                <WhcHigh>500kB</WhcHigh>
            </Watermarks>
        </Internal>
        <Tracing>
            <OutputFile>stdout</OutputFile>
    </Tracing>
    </Domain>
</CycloneDDS>
```

I see an error

```bash
async-std/: config: //CycloneDDS/Domain/General: Interfaces: unknown element (file:///etc/cyclone/cyclonedds.xml line 5)
```

### To reproduce

1. Use the below config on both machines (I have placed them under `/etc/cyclone/cyclonedds.xml`
```xml
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <Interfaces>
                <NetworkInterface name="lo" multicast="true" />
            </Interfaces>
            <AllowMulticast>default</AllowMulticast>
            <MaxMessageSize>65500B</MaxMessageSize>
        </General>
        <Internal>
            <Watermarks>
                <WhcHigh>500kB</WhcHigh>
            </Watermarks>
        </Internal>
        <Tracing>
            <OutputFile>stdout</OutputFile>
    </Tracing>
    </Domain>
</CycloneDDS>
```
2. In the bashrc set 
```bash
source /opt/ros/humble/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CYCLONEDDS_URI=file:///etc/cyclone/cyclonedds.xml
export CYCLONE_INCLUDE=/opt/ros/humble/include
export CYCLONE_LIB=/opt/ros/humble/lib/x86_64-linux-gnu
```
3. Let's say the IP of each machine is 192.168.50.201 and 192.168.50.202
4. On the machine with 201 run `ros2 run demo_nodes_cpp listener`
5. On the machine with 202 run `ros2 run demo_nodes_cpp talker`
6. On the machine with 201 run `ros2 run zenoh_bridge_dds zenoh_bridge_dds --no-multicast-scouting -l udp/0.0.0.0:7447`
7. On the machine with 202 run `ros2 run zenoh_bridge_dds zenoh_bridge_dds --no-multicast-scouting -l udp/0.0.0.0:7447 -e udp/192.168.50.201`

Both machines will give the error and won't communicate with each other

```bash
async-std/: config: //CycloneDDS/Domain/General: Interfaces: unknown element (file:///etc/cyclone/cyclonedds.xml line 5)
```

### System info

Platform: Ubuntu 22.04
ROS Distro: Humble
Zenoh: 0.5.0-3jammy.20221108.151336

JEnoch, 2023-04-19:
The version **0.5.0** of `zenoh-bridge-dds` you're using was embedding CycloneDDS 0.7.0.
At this time, the syntax it accepted for `NetworkInterface` configuration was:
`<NetworkInterfaceAddress>lo</NetworkInterfaceAddress>`

The new syntax you're using (`<NetworkInterface name="lo" multicast="true" />`) is accepted since `zenoh-bridge-dds` **0.7.0-rc** which embeds CycloneDDS 0.10.2.

Thus, you have 2 options:
 - The ideal would be for you to use `zenoh-bridge-dds` **0.7.0-rc**. It's not available as a ROS 2 package (the ROS buildfarm missed the appropriate Rust compiler to allow this). But you can easily install it as a pre-built binary following those instructions: https://github.com/eclipse-zenoh/zenoh-plugin-dds#how-to-install-it
 - You keep using `zenoh-bridge-dds` 0.5.0 as a ROS 2 package, but you define a 2nd `cyclonedds.xml` file for it using the old syntax for `<NetworkInterface>`. You can then override the `CYCLONEDDS_URI` environment variable when running your bridges as such:
  `CYCLONEDDS_URI=file:///etc/cyclone/cyclonedds_for_zenoh_bridge_0_5_0.xml`

sisaha9, 2023-04-20:
Specifying the CYCLONEDDS_URI file as a prefix worked for me. Thanks!

closed 2023-04-20 by sisaha9