eclipse-zenoh/zenoh-plugin-dds #153: [ Wrong Configuration ] Cannot stream pointcloud over Wifi

GiovanniBalestrieri, 2023-10-13:
### Describe the bug

I would like to stream pointcloud data from a host connected to a wired network to another one connected over wifi.
I am using a Zenoh Router and three Zenoh Bridges. The configuration works when all hosts are connected to the wired network. 

**Test 1 - Successfull **

Hosts A, B, C connected to the Ubuntu Router using ethernet cables.
[image]

**Test 2 - Failed **
Hosts A, B connected to the Ubuntu Router using ethernet cables. Host Connected over Wifi.
[image]

As soon as I run the zenoh-plugin-dds command. The connection freezes.

### To reproduce

I am using the following configuration in both cases. The wired case works well, the wireless doesn't.

1. Router: 
`zenohd`

2. Host A:
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`
ros2 launch ros_package_x publish_string_topic.launch.py

2. Host B:
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`
ros2 launch ros_package_x publish_pointcloud_topic.launch.py

2. Host C
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`
ros2 run rviz2 rviz2

### System info

- Router
Ubuntu 20.04 
IP: 192.168.0.1
zenoh version 0.10
zenoh command:
`zenohd `

- Host A
Ubuntu 20.04
IP: 192.168.0.2
Ros2 Foxy and CycloneDDS RMW
zenoh version 0.10
zenoh command:
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`

- Host B
Ubuntu 20.04
IP: 192.168.0.3
Ros2 Foxy and CycloneDDS RMW
zenoh version 0.10
zenoh command:
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`

- Host C - Wired
Ubuntu 20.04
IP: 192.168.0.5
Ros2 Foxy and CycloneDDS RMW
zenoh version 0.10
zenoh command:
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`

- Host C - Wifi
Ubuntu 20.04
IP: 192.168.0.226
Ros2 Foxy and CycloneDDS RMW
zenoh version 0.10
zenoh command:
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 --no-multicast-scouting`

kydos, 2023-10-13:
Hello @GiovanniBalestrieri, looking at the configuration above the issues seems to be that you are always passing the same locator: `tcp/192.168.0.1:7447`. This would work as a point of RDV if you would enable the gossip discovery (see config example on zenoh distro). The other approach is to pass the locators of the other machines. 

HTH.

GiovanniBalestrieri, 2023-10-17:
Hi @kydos, I am happy to modify the approach. I took a bit of time to further study the documentation. ESpecially [this](https://conferences2.sigcomm.org/acm-icn/2022/assets/zenoh-4-Zenoh-and-Zenoh-Flow-Hands-on-e8cbd760e0b88b74417fb1c14d1d373b5ce2a094bc29b5f1a0bfd8e52030c151.pdf) pdf and the help section of zenohd 

```
-l, --listen <ENDPOINT>...
A locator on which this router will listen for
incoming sessions. Repeat this option to open
several listeners.
-e, --connect <ENDPOINT>...
A peer locator this router will try to
connect to. Repeat this option to connect to
several peers.
```

> _The other approach is to pass the locators of the other machines._

**Attempt I Wired setup**
I tried passing to Host A and B the locator each other using
- Router - 192.168.0.1
```  zenohd```

- Host A - IP: 192.168.0.2

> zenoh-bridge-dds -m peer -e tcp/192.168.0.1:7447 -l tcp/192.168.0.3:7447
> [2023-10-17T12:40:14Z INFO  zenoh_bridge_dds] zenoh-bridge-dds v0.10.0-rc built with rustc 1.72.1 (d5c2e9c34 2023-09-13)
> [2023-10-17T12:40:14Z INFO  zenoh::net::runtime] Using PID: c944286c0f310276499cf9908db65b89
> [2023-10-17T12:40:14Z ERROR zenoh::net::runtime::orchestrator] Unable to open listener tcp/192.168.0.3:7447: Can not create a new TCP listener bound to tcp/192.168.0.3:7447: [192.168.0.3:7447: Cannot assign requested address (os error 99) at /cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-link-tcp-0.10.0-rc/src/unicast.rs:245.] at /cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-link-tcp-0.10.0-rc/src/unicast.rs:333.
> thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Can not create a new TCP listener bound to tcp/192.168.0.3:7447: [192.168.0.3:7447: Cannot assign requested address (os error 99) at /cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-link-tcp-0.10.0-rc/src/unicast.rs:245.] at /cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-link-tcp-0.10.0-rc/src/unicast.rs:333.', zenoh-bridge-dds/src/main.rs:247:62
> stack backtrace:
>    0:       0x558887641c - <unknown>
>    1:       0x558869c0b0 - <unknown>
>    2:       0x558884b740 - <unknown>
>    3:       0x5588877484 - <unknown>
>    4:       0x5588877090 - <unknown>
>    5:       0x5588877fa4 - <unknown>
>    6:       0x5588877af0 - <unknown>
>    7:       0x5588877a64 - <unknown>
>    8:       0x5588877a58 - <unknown>
>    9:       0x558857e9ec - <unknown>
>   10:       0x558857ecfc - <unknown>
>   11:       0x55886274e4 - <unknown>
>   12:       0x558861ca6c - <unknown>
>   13:       0x55885c7ee0 - <unknown>
>   14:       0x5588627c9c - <unknown>
>   15:       0x7fa54c2e10 - __libc_start_main
>   16:       0x55885afda4 - <unknown>
> Aborted

- Host B - IP: 192.168.0.3
   zenoh-bridge-dds -m peer -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447
   Same error message as above

Why am I getting this error? It will help me understand what is happening under the hood. 

**Attempt II Wired setup**
Using client mode 

- Router - 192.168.0.1
```  zenohd```

- Host A - IP: 192.168.0.2
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.3:7447 -l tcp/192.168.0.5:7447`

- Host B - IP: 192.168.0.3
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.5:7447`

- Host C - IP: 192.168.0.5
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.3:7447`

No error  messages. 
Opened Rviz2 on host C, one poincloud message got through then the connection to Host B, the one publishing the pointcloud, dropped for a few seconds. 

**Attempt III Wired setup --no-multicast-scouting**
Using client mode 

- Router - 192.168.0.1
```  zenohd```

- Host A - IP: 192.168.0.2
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.3:7447 -l tcp/192.168.0.5:7447` --no-multicast-scouting

- Host B - IP: 192.168.0.3
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.5:7447` --no-multicast-scouting

- Host C - IP: 192.168.0.5
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.3:7447  --no-multicast-scouting`

No error  messages. 
Opened Rviz2 on host C and got 18 messages per second. Ok.

**Attempt IV Wireless setup --no-multicast-scouting**
Moving to the wireless case. 

- Router - 192.168.0.1
```  zenohd```

- Host A - IP: 192.168.0.2
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.3:7447 -l tcp/192.168.0.5:7447` --no-multicast-scouting

- Host B - IP: 192.168.0.3
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.5:7447` --no-multicast-scouting

- Host C - IP: 192.168.0.5
`zenoh-bridge-dds -m client -e tcp/192.168.0.1:7447 -l tcp/192.168.0.2:7447 -l tcp/192.168.0.3:7447  --no-multicast-scouting`

GiovanniBalestrieri, 2023-10-17:
Now reading these resources to better understand the situation
- https://zenoh.io/docs/getting-started/deployment/
- https://www.youtube.com/watch?v=ahO4kT_Zg7s&t=1577s

JEnoch, 2023-11-28:
Hi @GiovanniBalestrieri ,

We recently developed a new Zenoh bridge for ROS 2:  https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds
I recommand you switch to this one as it better integrates with ROS 2 and brings these benefits:

- Better integration of the ROS graph (all ROS topics/services/actions can be seen across bridges)
- Better support of ROS toolings (ros2 CLI, rviz2...)
- Configuration of a ROS namespace on the bridge (instead of on each ROS Node)
- Services and Action as Zenoh Queryables with more efficiency and scalability that RPC over DDS
- Even more compact discovery information between the bridges (not forwarding all ros_discovery_info messages as such)

RViz and the ros2 commands are working smoothly with it, as I showed in a demo at ROSCon 2023.

Still, publishing at high rate larges data such as pointcloud might be problematic over WiFi where bandwidth might be limited and messages loss could occur.
My advise is also to lower the frequency of republication by the bridge for your pointcloud topic using the [`pub_max_frequency`](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/a485c63840b2be5832eb0ffabc98e75874b79943/DEFAULT_CONFIG.json5#L81) configuration.

I'm closing this issue since I hope you'll test the new bridge with this config. If you still have troubles, please open a new issue in [its repo](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues).

closed 2023-11-28 by JEnoch