eclipse-zenoh/zenoh-plugin-dds #157: [Bug] Cannot work with FastDDS.

YXL76, 2023-11-06:
### Describe the bug

Cannot route ros2 topic.

### To reproduce

Terminal one:

```bash
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
ros2 topic pub /chatter std_msgs/Empty
```

Terminal two:

```plain
$ RUST_LOG=debug ./zenoh-bridge-dds
[2023-11-06T09:10:51Z INFO  zenoh_bridge_dds] zenoh-bridge-dds v0.10.0-rc built with rustc 1.72.0 (5680fa18f 2023-08-23)
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime] Zenoh Rust API v0.10.0-rc
[2023-11-06T09:10:51Z INFO  zenoh::net::runtime] Using PID: 41126de29723b95214b4535c2fa37bb7
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::network] [Routers network] Add node (self) 41126de29723b95214b4535c2fa37bb7
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::network] [Peers network] Add node (self) 41126de29723b95214b4535c2fa37bb7
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime::orchestrator] Listener added: tcp/[::]:33463
[2023-11-06T09:10:51Z INFO  zenoh::net::runtime::orchestrator] Zenoh can be reached at: tcp/100.122.124.205:33463
[2023-11-06T09:10:51Z INFO  zenoh::net::runtime::orchestrator] Zenoh can be reached at: tcp/[fe80::8a66:64ff:fe7a:7ccd]:33463
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime::orchestrator] UDP port bound to 224.0.0.224:7446
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime::orchestrator] Joined multicast group 224.0.0.224 on interface 100.122.124.205
[2023-11-06T09:10:51Z INFO  zenoh::net::runtime::orchestrator] zenohd listening scout messages on 224.0.0.224:7446
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime::orchestrator] UDP port bound to 100.122.124.205:37112
[2023-11-06T09:10:51Z DEBUG zenoh::net::runtime::orchestrator] Waiting for UDP datagram...
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] DDS plugin v0.10.0-rc built with rustc 1.72.0 (5680fa18f 2023-08-23)
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] DDS plugin Config { scope: None, domain: 0, group_member_id: None, allow: None, deny: None, max_frequencies: [], generalise_subs: [], generalise_pubs: [], forward_discovery: false, reliable_routes_blocking: true, localhost_only: false, queries_timeout: 5s, __required__: false, __path__: [] }
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::router] New Face{0, 41126de29723b95214b4535c2fa37bb7}
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::pubsub] Register client subscription
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::resource] Register resource @/liveliness/zenoh-plugin-dds/41126de29723b95214b4535c2fa37bb7
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::pubsub] Register subscription @/liveliness/zenoh-plugin-dds/41126de29723b95214b4535c2fa37bb7 for Face{0, 41126de29723b95214b4535c2fa37bb7}
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] Create DDS Participant with CYCLONEDDS_URI=''
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] DDS plugin 41126de29723b95214b4535c2fa37bb7 with member_id=41126de29723b95214b4535c2fa37bb7 and using DDS Participant 01102f9c7367a1a7fd9c9931000001c1
[2023-11-06T09:10:51Z DEBUG zenoh_ext::querying_subscriber] Fetch
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::queries] Route query Face{0, 41126de29723b95214b4535c2fa37bb7}:0 for res @/liveliness/zenoh-plugin-dds/**
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::queries] Send final reply Face{0, 41126de29723b95214b4535c2fa37bb7}:0 (no matching queryables or not master)
[2023-11-06T09:10:51Z DEBUG zenoh_ext::querying_subscriber] All fetches done. Replies and live publications merged - 1 samples to propagate
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] Declare admin space on @/service/41126de29723b95214b4535c2fa37bb7/dds/**
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::queries] Register client queryable @/service/41126de29723b95214b4535c2fa37bb7/dds/**
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::resource] Register resource @/service/41126de29723b95214b4535c2fa37bb7/dds/**
[2023-11-06T09:10:51Z DEBUG zenoh::net::routing::queries] Register queryable @/service/41126de29723b95214b4535c2fa37bb7/dds/** (face: Face{0, 41126de29723b95214b4535c2fa37bb7})
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] Run in "local discovery" mode
[2023-11-06T09:10:51Z DEBUG zenoh_plugin_dds] New zenoh_dds_plugin detected: 41126de29723b95214b4535c2fa37bb7
[2023-11-06T09:10:52Z DEBUG zenoh_plugin_dds::dds_mgt] Discovered DDS Participant 010f7e166410877201000000000001c1)
[2023-11-06T09:10:52Z DEBUG zenoh_plugin_dds] Discovered DDS Participant 010f7e166410877201000000000001c1
[2023-11-06T09:10:53Z DEBUG zenoh_plugin_dds::dds_mgt] Discovered DDS Participant 010f7e16b44544b701000000000001c1)
[2023-11-06T09:10:53Z DEBUG zenoh_plugin_dds] Discovered DDS Participant 010f7e16b44544b701000000000001c1
```

There is no `Route DDS->Zenoh (rt/chatter -> rt/chatter): created with topic_type=std_msgs::msg::dds_::Empty_` in the output.

### System info

- Platform: Ubuntu 20.04 64-bit
- Zenoh: 0.10.0-rc
- ROS: foxy

samcarey, 2023-11-28:
@YXL76 Do other values for `RMW_IMPLEMENTATION` work other than `rmw_fastrtps_cpp`?

I'm testing this DDS plugin (master, latest) with the [FastDDS Shapes Demo](https://www.eprosima.com/index.php/products-all/eprosima-shapes-demo) and getting the same result (sees DDS Participant, but not the writer).

Incidentally, I had to create an issue for [DustDDS](https://github.com/s2e-systems/dust-dds/issues/176) when I found that that it also incompatible with FastDDS. I wonder if this Zenoh bridge is suffering from the same issue.

closed 2023-11-28 by YXL76

JEnoch, 2023-11-28:
@YXL76 : as you closed the ticket, did you found the cause of the issue and a solution ?
I tried to reproduce it with Rolling and Humble, but my test was working as expected:
- terminal 1: `RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 topic pub /chatter std_msgs/Empty`
- terminal 2: `zenoh-bridge-dds  -f`
- terminal 3: `zenoh-bridge-dds  -f -d 1`
- terminal 4: `RMW_IMPLEMENTATION=rmw_fastrtps_cpp ROS_DOMAIN_ID=1 ros2 topic echo /chatter`

Maybe the issue was in Foxy's `rmw_fastrtps_cpp` and has been fixed in Humble ?

@samcarey : I reproduce the issue with FastDDS ShapesDemo. However, looking at DDS network traffic with Wireshark, I see ShapesDemo is sending DATA(P) messages (SPDP for Participant discovery) along with non-standard submessages (id=0x80), but no DATA(W) nor DATA(R) (SEDP for Reader/Writer discovery). 
 I'm not familiar with FastDDS configuration, but I guess that something makes it behave in a non-standard way, not sending discovery messages for Publishers, Subscribers nor Topics.