eclipse-zenoh/zenoh-plugin-dds #164: [Bug]  Iron and Rolling: errors on ros_discovery_info messages

JEnoch, 2023-11-29:
### Describe the bug

Following https://github.com/ros2/rmw_dds_common/pull/68 the `Gid` type used in `ros_discovery_info` topic changed from `char[24] data` to `char[16] data`.

This leads the bridge to be incompatible with Iron and latest Rolling.
The incompatibility can be seen with such logs in the bridge:
`[2023-11-29T14:51:27Z WARN  zenoh_plugin_ros2dds::ros_discovery] Error receiving ParticipantEntitiesInfo on ros_discovery_info: invalid utf-8 sequence of 1 bytes from index 4` 

Or in a ROS Node:
`[WARN] [1701269487.404903002] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.`

### To reproduce

In Iron, run `zenoh-bridge-dds -f` with any ROS Node and see the logs.

### System info

- commit: [55a1385](https://github.com/eclipse-zenoh/zenoh-plugin-dds/commit/55a13855ab0d698b0e7cd078588950d867de9d99)
- ROS distro: Iron and Rolling
- Platform: all

Timple, 2024-03-04:
Is there any update here? On ROS Iron I'm still getting:
``` 
[rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/tf' with type 'tf2_msgs::msg::dds_::TFMessage_' from USER_DATA '(null)'.
```

JEnoch, 2024-03-04:
I don't think the error message you get is related to this issue.
This issue is specifically related to the `ros_discovery_info` which is published and subscribed and decoded by the bridge.
You error message is related to the `tf` topic which the bridge is routing un-touched.
Therefore I think more of an incompatibility between your ROS Node publishing on `tf` topic and your ROS Node subscribing to this same topic.

Wrt. this issue, it has already been fixed for the [`zenoh-plugin-ros2dds`](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds): https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/21 
And we advise all ROS 2 users to use this new `zenoh-bridge-ros2dds` instead of the `zenoh-bridge-dds`.
So I think we will fix this issue only if some ROS 2 user have good reasons to not switch to the new bridge.

Timple, 2024-03-04:
Ah, we are actually using the `zenoh-plugin-ros2dds`. So my issue must indeed be different. Sorry for the confusion.