eclipse-zenoh/zenoh-plugin-dds #279: [Bug] Massive performance drop on lateast release 1.0.0-alpha.4

franz6ko, 2024-07-17:
### Describe the bug

This is related to this other bug in zenoh_plugin_ros2dds [here](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/190).

The solution for me was the same, rolling back to a previous version. In this case 0.10.0-rc-1

### To reproduce

Updated to last version.
Check rate of high speed messages (previously 1000Hz dropped to 200Hz)

### System info

Ubuntu 20.04

JEnoch, 2024-07-17:
Thanks for this feedback.
Could you please detail the scenario in which you see such performance drop ?
Is it some kind of throughput test ? With which size of payload ?

franz6ko, 2024-07-17:
Hi @JEnoch,

Yes, the test was the same I explain in [this other issue ](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/192) but in this case between the release 1.0.0-alpha.4 and the 0.10.0-rc-1.

In this case, a 1000Hz topic will drop to around 200Hz on 1.0.0-alpha.4 but would stay at 1000Hz on 0.10.0-rc-1.

I'm using it with ROS2 Foxy on Ubuntu 20.04. The topic message is basic structure holding some floats.

JEnoch, 2024-07-19:
Probably the same cause than explained in https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/192#issuecomment-2239055460 .
The issue is caused by some weird interaction between `tokio` and CycloneDDS. 
Investigation in progress...

JEnoch, 2024-08-28:
Zenoh has now been fixed (in branch `dev/1.0.0`) to adapt its batching mechanism to the `tokio` behaviour:
https://github.com/eclipse-zenoh/zenoh/pull/1335  
I can confirm that with default configuration (i.e. without configuring the `express` flag for publishers) the routing of 1kHz publications is no longer suffering performance drop.

This repo will be sync to use the fixed Zenoh version in branch `dev/1.0.0` in the next few days.

closed 2024-08-28 by JEnoch

franz6ko, 2024-08-28:
Great to hear!! Thanks for your support. I'm happy to have contributed to this issue.