eclipse-zenoh/zenoh-plugin-dds #346: [Bug] Crash with Cyclone 0.10.5 ping pong examples

gabrik, 2024-09-23:
### Describe the bug

The pluing crashes on discovery of ping/pong participants.

```
zenoh-1  | 2024-09-18T12:29:00.478812Z  INFO main ThreadId(01) zenohd: zenohd v9bc441f-modified built with rustc 1.81.0 .....
zenoh-1  | 2024-09-18T12:29:00.538520Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds::route_zenoh_dds: Route Zenoh->DDS (ping/RoundTrip -> RoundTrip): creation with topic_type=RoundTripModule::DataType querying_subscriber=false
zenoh-1  | 2024-09-18T12:29:00.538541Z DEBUG                 main ThreadId(01) zenoh::net::runtime::orchestrator: Joined multicast group 224.0.0.224 on interface 172.18.0.2
zenoh-1  | 2024-09-18T12:29:00.538551Z  INFO                 main ThreadId(01) zenoh::net::runtime::orchestrator: zenohd listening scout messages on 224.0.0.224:7446
zenoh-1  | 2024-09-18T12:29:00.538565Z DEBUG tokio-runtime-worker ThreadId(59) zenoh::net::routing::dispatcher::pubsub: Face{1, dd5} Declare subscriber 13 (ping/RoundTrip)
zenoh-1  | thread '<unnamed>' panicked at /usr/local/cargo/git/checkouts/zenoh-plugin-dds-89876188164765d1/5cc50d6/zenoh-plugin-dds/src/dds_mgt.rs:135:28:
zenoh-1  | misaligned pointer dereference: address must be a multiple of 0x8 but is 0x800000004
zenoh-1  | stack backtrace:
zenoh-1  | 2024-09-18T12:29:00.538608Z DEBUG tokio-runtime-worker ThreadId(59) zenoh::net::routing::dispatcher::resource: Register resource ping/RoundTrip
zenoh-1  | 2024-09-18T12:29:00.538692Z DEBUG                 main ThreadId(01) zenoh::net::runtime::orchestrator: UDP port bound to 172.18.0.2:56918
zenoh-1  | 2024-09-18T12:29:00.538781Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds::route_zenoh_dds: Route Zenoh->DDS (ping/RoundTrip -> RoundTrip): create DDS Writer
zenoh-1  | 2024-09-18T12:29:00.538820Z DEBUG                net-0 ThreadId(67) zenoh::net::runtime::orchestrator: Waiting for UDP datagram...
zenoh-1  | 2024-09-18T12:29:00.538975Z  INFO tokio-runtime-worker ThreadId(59) zenoh_plugin_dds: Route Zenoh->DDS (ping/RoundTrip -> RoundTrip): created with topic_type=RoundTripModule::DataType
zenoh-1  | 2024-09-18T12:29:00.539135Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds: Discovered DDS Writer 0110356d96aa96de8aca75ee00000203 on RoundTrip with type 'RoundTripModule::DataType' and QoS: Qos { user_data: None, topic_data: None, group_data: None, durability: None, durability_service: None, presentation: None, deadline: None, latency_budget: None, ownership: None, ownership_strength: None, liveliness: None, time_based_filter: None, partition: Some(["ping"]), reliability: Some(Reliability { kind: RELIABLE, max_blocking_time: 10000000000 }), transport_priority: None, lifespan: None, destination_order: None, history: None, resource_limits: None, writer_data_lifecycle: Some(WriterDataLifecycle { autodispose_unregistered_instances: false }), reader_data_lifecycle: None, writer_batching: None, type_consistency: None, entity_name: None, properties: None, ignore_local: None, data_representation: Some([0, 2]) }
zenoh-1  | 2024-09-18T12:29:00.539226Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds::route_dds_zenoh: Route DDS->Zenoh (RoundTrip -> ping/RoundTrip): creation with topic_type=RoundTripModule::DataType
zenoh-1  | 2024-09-18T12:29:00.539432Z DEBUG tokio-runtime-worker ThreadId(59) zenoh::net::routing::dispatcher::interests: Face{1, dd5} Declare interest 14 (ping/RoundTrip)
zenoh-1  | 2024-09-18T12:29:00.539501Z DEBUG tokio-runtime-worker ThreadId(59) zenoh::net::routing::dispatcher::interests: Face{1, dd5} Undeclare interest 14
zenoh-1  | 2024-09-18T12:29:00.539783Z  INFO tokio-runtime-worker ThreadId(59) zenoh_plugin_dds: Route DDS->Zenoh (RoundTrip -> ping/RoundTrip): created with topic_type=RoundTripModule::DataType
zenoh-1  | 2024-09-18T12:29:00.539853Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds: Discovered DDS Reader 0110356d96aa96de8aca75ee00000304 on RoundTrip with type 'RoundTripModule::DataType' and QoS: Qos { user_data: None, topic_data: None, group_data: None, durability: None, durability_service: None, presentation: None, deadline: None, latency_budget: None, ownership: None, ownership_strength: None, liveliness: None, time_based_filter: None, partition: Some(["pong"]), reliability: Some(Reliability { kind: RELIABLE, max_blocking_time: 10000000000 }), transport_priority: None, lifespan: None, destination_order: None, history: None, resource_limits: None, writer_data_lifecycle: None, reader_data_lifecycle: None, writer_batching: None, type_consistency: Some(TypeConsistency { kind: ALLOW_TYPE_COERCION, ignore_sequence_bounds: true, ignore_string_bounds: true, ignore_member_names: false, prevent_type_widening: false, force_type_validation: false }), entity_name: None, properties: None, ignore_local: None, data_representation: Some([0, 2]) }
zenoh-1  | 2024-09-18T12:29:00.539909Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds::route_zenoh_dds: Route Zenoh->DDS (pong/RoundTrip -> RoundTrip): creation with topic_type=RoundTripModule::DataType querying_subscriber=false
zenoh-1  | 2024-09-18T12:29:00.539951Z DEBUG tokio-runtime-worker ThreadId(59) zenoh::net::routing::dispatcher::pubsub: Face{1, dd5} Declare subscriber 15 (pong/RoundTrip)
zenoh-1  | 2024-09-18T12:29:00.540068Z DEBUG tokio-runtime-worker ThreadId(59) zenoh_plugin_dds::route_zenoh_dds: Route Zenoh->DDS (pong/RoundTrip -> RoundTrip): create DDS Writer
zenoh-1  | 2024-09-18T12:29:00.540162Z  INFO tokio-runtime-worker ThreadId(59) zenoh_plugin_dds: Route Zenoh->DDS (pong/RoundTrip -> RoundTrip): created with topic_type=RoundTripModule::DataType
zenoh-1  |    0:     0x558ceada50b5 - std::backtrace_rs::backtrace::libunwind::trace::h1b4b7c200f0b1134
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/../../backtrace/src/backtrace/libunwind.rs:116:5
zenoh-1  |    1:     0x558ceada50b5 - std::backtrace_rs::backtrace::trace_unsynchronized::h40ba1b1a720cbf98
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
zenoh-1  |    2:     0x558ceada50b5 - std::sys::backtrace::_print_fmt::h8992bf724ef0f4bd
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/sys/backtrace.rs:65:5
zenoh-1  |    3:     0x558ceada50b5 - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::h358afad87e02ca76
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/sys/backtrace.rs:40:26
zenoh-1  |    4:     0x558ceadd26bb - core::fmt::rt::Argument::fmt::h414c419d4b9c8d82
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/core/src/fmt/rt.rs:173:76
zenoh-1  |    5:     0x558ceadd26bb - core::fmt::write::hb19b5b269a2fe458
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/core/src/fmt/mod.rs:1182:21
zenoh-1  |    6:     0x558ceada0aff - std::io::Write::write_fmt::he5a92676a45ef09d
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/io/mod.rs:1827:15
zenoh-1  |    7:     0x558ceada63a1 - std::sys::backtrace::BacktraceLock::print::h6d30d1c1b5775240
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/sys/backtrace.rs:43:9
zenoh-1  |    8:     0x558ceada63a1 - std::panicking::default_hook::{{closure}}::h3bff550b24d93725
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/panicking.rs:269:22
zenoh-1  |    9:     0x558ceada607c - std::panicking::default_hook::hd53b1b06d2b99687
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/panicking.rs:296:9
zenoh-1  |   10:     0x558ceada6a01 - std::panicking::rust_panic_with_hook::h9fdd87cddb2763da
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/panicking.rs:800:13
zenoh-1  |   11:     0x558ceada6867 - std::panicking::begin_panic_handler::{{closure}}::h089783ab6b5cba45
zenoh-1  |                                at /rustc/eeb90cda1969383f56a2637cbd3037bdf598841c/library/std/src/panicking.rs:674:13
zenoh-1  |   12:     0x558ceada5579 - std::sys::backtrace::__rust_end_short_backtrace::hed34776d77ef7922
[... cut ...]
zenoh-1  |                                at /home/buildozer/aports/main/musl/src/1.2.4/src/thread/pthread_create.c:207:2
zenoh-1  | thread caused non-unwinding panic. aborting.
```

Cause seems:

```
zenoh-1  | thread '<unnamed>' panicked at /usr/local/cargo/git/checkouts/zenoh-plugin-dds-89876188164765d1/5cc50d6/zenoh-plugin-dds/src/dds_mgt.rs:135:28:
zenoh-1  | misaligned pointer dereference: address must be a multiple of 0x8 but is 0x800000004
```

### To reproduce

Start the bridge
Start ping pong exampls

### System info

All

gmartin82, 2024-09-25:
@gabrik was the binary built with support for the 'zenoh-plugin-ros2dds' included?

The ROS2 and DDS plugins use different versions of the cyclors library and it looks like when a binary is built with both of these plugins included there is a clash between the versions. In the stack trace it can be seen that cyclors 0.2.1 is being invoked whereas the DDS plugin should be using cyclors 0.3.1.

gabrik, 2024-09-25:
Yes, that's the case.
I wonder how this happens as rust builds both version of `cyclors` and it should be able to link both versions.

gmartin82, 2024-09-25:
In this case, Cyclone DDS is calling back to Rust when a sample is received. The issue is possibly caused by combining multiple versions of the Cyclone C library into a single binary.

gabrik, 2024-09-25:
Yeap but:

```
$ cargo tree -i cyclors --features plugins
error: There are multiple `cyclors` packages in your project, and the specification `cyclors` is ambiguous.
Please re-run this command with one of the following specifications:
  cyclors@0.2.1
  cyclors@0.3.1
```

So rust does know that there are 2 different version, and know which plugin belongs to which version

```
$ cargo tree -i cyclors@0.3.1 --features plugins
cyclors v0.3.1
└── zenoh-plugin-dds v1.0.0-dev (https://github.com/eclipse-zenoh/zenoh-plugin-dds?branch=main#6a12ef01)
    └── zetta-uservice-topology v1.0.0 (/home/gabrik/Workspace/zettascale/paas/zetta-backend/zetta-uservice-topology)
```
```
$ cargo tree -i cyclors@0.2.1 --features plugins
cyclors v0.2.1
└── zenoh-plugin-ros2dds v1.0.0-dev (https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds?branch=main#1e53a50b)
    └── zetta-uservice-topology v1.0.0 (/home/gabrik/Workspace/zettascale/paas/zetta-backend/zetta-uservice-topology)
```

the only way the DDS plugin can call the "wrong" `cyclors` is if the underlying C functions are "overwritten" at compile time by the old ones.

Cant's `cyclors` "mangle" the names including the version? This would prevent this sort of issues.
Or maybe something needs to be setup in cargo configuration

gmartin82, 2024-09-25:
Yes, I think C functions are likely being "overwritten" during linking.

It might be possible to "mangle" the Cyclone symbols (possibly by adding a prefix) to make them unique for each version of Cyclors. This is something that might be worth investigating if we do need to use multiple versions of Cyclone at once.

gabrik, 2024-10-29:
Solved by #376

closed 2024-10-29 by gabrik