eclipse-zenoh/zenoh-plugin-dds #402: [Bug] Unable to Receive Messages on Subscriber Node via Zenoh Router deployed in Cloud

Jianqiao94, 2024-12-01:
### Describe the bug

I am experiencing an issue where I cannot receive messages on a listener node across a Zenoh router deployed on a cloud server. The setup involves two devices running ROS2 Humble and zenoh-bridge-ros2dds. The issue is that, while the topic /talker/chatter is correctly discovered on the listener node (ros2 topic list shows the topic), ros2 topic echo /talker/chatter does not display any messages.

### To reproduce

1. Set up the zenoh router on cloud server with public IP. 
**./zenohd --listen tcp/0.0.0.0:7447 --config <router_config.json5>**
I use default configuration, just set mode to router. 

2. Talker/listener node configuration
- Install ROS2 Humble and zenoh-bridge-ros2dds.
- Configure file something like:
**{
  "plugins": {
    "ros2dds": {
      "namespace": "/talker",
      "domain": 0,
      "ros_localhost_only": false,
      "deny": {
        "publishers": [".*/rosout", ".*/parameter.*"],
        "subscribers": [".*/rosout", ".*/parameter.*"]
      }
    }
  },
  "connect": {
    "endpoints": [
      "tcp/<public_ip_of_router>:7447"
    ]
  }
}**
3. Start zenoh router, zenoh bridge and talker/listener node
4. On the listener node, the log says something like 
_"2024-11-28T20:44:29.966517Z  INFO main ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.8.163:36167
2024-11-28T20:44:30.035564Z  INFO tokio-runtime-worker ThreadId(18) zenoh_plugin_ros2dds: Remote bridge <ZID> announces Publisher talker/chatter
2024-11-28T20:44:30.036093Z  INFO tokio-runtime-worker ThreadId(18) zenoh_plugin_ros2dds::routes_mgr: Route Subscriber (Zenoh:listener/talker/chatter -> ROS:/talker/chatter) created'_  
**So I think the topic is indeed discovered but I cannot view it or echo it.** 

### System info

Ubuntu 22.04, ROS2 humble.

rahulkatiyar19955, 2025-04-08:
I m facing the same problem.

Jianqiao94, 2025-04-08:
> I m facing the same problem.

hi mate，I solved this problem in the end by not using bridge but use zenoh RMW directly and it works

rahulkatiyar19955, 2025-04-11:
In my usecase, it is not possible to change the robot code as it private to a client, we can only provide a wrapper which i was doing with a isolated docker of zenoh_dds_router.