eclipse-zenoh/zenoh-plugin-dds #516: [Bug] Segmentation fault when paired with RTI Persistence Service and Dispose on Keyed Objects

seichmann, 2025-05-16:
### Describe the bug

When RTI Persistence Service (Version 7.3.0) is inside the DDS Domain, whenever a DDS Object is being disposed, the Zenoh Process crashes completely with _"2573439 Segmentation fault."_

### To reproduce

1. Start RTI Persistence Service
2. Start Zenoh
3. Configure QOS to DDS_PERSISTENT_DURABILITY_QOS
4. Create a keyed object inside the DDS Domain via a DDSWriter
5. Delete/Dispose the keyed object inside the DDS Domain via a DDSWriter
6. Zenoh craches

### System info

- Platform: Redhat 9
- Zenoh: 1.2.1-1.3.4  
- RTI DDS 7.3.0

gmartin82, 2025-05-16:
@seichmann we don't have access to RTI software. Could you provide more information, such as the information from a core dump, please?

seichmann, 2025-05-19:
Dear Martin,

could you explain what exactly do you need ?

Here is the output of coredumpctl debug 2544952:

           PID: 2544952 (zenohd)
           UID: 2036 (eichmann)
           GID: 10002 (iness-admin-tooling)
        Signal: 11 (SEGV)
     Timestamp: Fri 2025-05-16 12:10:36 CEST (2 days ago)
  Command Line: ./zenohd --config zenoh1.json5
    Executable: /Development/zenoh-1.2.1-own-build/zenohd
 Control Group: /user.slice/user-2036.slice/user@2036.service/app.slice/app-org.gnome.Terminal.slice/vte-spawn-36fe12fb-498b-4398-ab26-06e6fb7a1973.scope
          Unit: user@2036.service
     User Unit: vte-spawn-36fe12fb-498b-4398-ab26-06e6fb7a1973.scope
         Slice: user-2036.slice
     Owner UID: 2036 (eichmann)
       Boot ID: 92aa03cf05de4ca3a191d0a28a446775
    Machine ID: 1b14cfd91ca546aabd670bc304722f17
      Hostname: xxxx
       Storage: /var/lib/systemd/coredump/core.zenohd.2036.92aa03cf05de4ca3a191d0a28a446775.2544952.1747390236000000.zst (present)
  Size on Disk: 779.0K
       Message: Process 2544952 (zenohd) of user 2036 dumped core.
                
                Stack trace of thread 2544963:
                #0  0x0000000000000000 n/a (n/a + 0x0)
                #1  0x00007f9dd9304300 pregrams (libzenoh_plugin_dds.so + 0x504300)
                #2  0x0000000000000000 n/a (n/a + 0x0)
                ELF object binary architecture: AMD x86-64

GNU gdb (Red Hat Enterprise Linux) 14.2-3.el9
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Development/zenoh-1.2.1-own-build/zenohd...
[New LWP 2544963]
[New LWP 2544953]
[New LWP 2544952]
[New LWP 2544955]
[New LWP 2544954]
[New LWP 2544957]
[New LWP 2544956]
[New LWP 2544958]
[New LWP 2544960]
[New LWP 2544961]
[New LWP 2544959]
[New LWP 2544969]
[New LWP 2544964]
[New LWP 2544971]
[New LWP 2544962]
[New LWP 2544966]
[New LWP 2544970]
[New LWP 2544965]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Core was generated by `./zenohd --config zenoh1.json5'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000000000000000 in ?? ()
[Current thread is 1 (Thread 0x7f9dbe7fc640 (LWP 2544963))]
warning: Missing auto-load script at offset 0 in section .debug_gdb_scripts
of file /Development/zenoh-1.2.1-own-build/zenohd.

Regards
Stephan

seichmann, 2025-05-19:
[zenoh_trace.log](https://github.com/user-attachments/files/20279164/zenoh_trace.log)

gmartin82, 2025-05-19:
A backtrace from GDB (`bt`) of the core dump might be useful to show what was happening at the time.

Compiling the plugin in debug mode (`cargo build` rather than `cargo build --release`) may provide more information as well if your not already compiling this way.

seichmann, 2025-05-19:
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007f9dd917ca58 in remote_make_sample () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#2  0x00007f9dd914b110 in ddsi_deliver_locally_allinsync () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#3  0x00007f9dd917cebd in deliver_user_data () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#4  0x00007f9dd917e0d9 in handle_regular () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#5  0x00007f9dd9181e81 in handle_submsg_sequence.constprop.0.isra () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#6  0x00007f9dd91834dc in do_packet () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#7  0x00007f9dd9183c83 in ddsi_recv_thread () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#8  0x00007f9dd9184c27 in create_thread_wrapper () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#9  0x00007f9dd913453d in os_startRoutineWrapper () from /Development/zenoh-1.2.1-own-build/libzenoh_plugin_dds.so
#10 0x00007f9dd98897e2 in start_thread () from /lib64/libc.so.6
#11 0x00007f9dd990e800 in clone3 () from /lib64/libc.so.6

It should have been compiled in debug mode. Seems inside the cyclors dependency...

seichmann, 2025-05-19:
I narrowed down the issue most probably to the mechanism around serialize_key_with_dispose: https://community.rti.com/static/documentation/connext-dds/current/doc/manuals/connext_dds_professional/users_manual/users_manual/DATA_WRITER_PROTOCOL_Qos.htm#sending_2410472787_2038359

Let me explain the behavior, as it is not related only to RTI Persistence Service, it also happens when you have a regular DDS Writer with serialize_key_with_dispose=false. I also found some hints about potential incompatibility here https://cyclonedds.io/docs/cyclonedds/latest/config/conformance.html. In our case, our key is in practice a UUID:

   struct UUID
        {
            long long low;
            long long high;
        };

1) If serialize_key_with_dispose=true, Zenoh can handle the DDS Message, but ignores it.
2) If serialize_key_with_dispose=false, Zenoh breaks.

As written in the RTI documentation, the RTI Persistence Service ignores always the serialize_key_with_dispose configuration, so thats why it is always on the second case.

Does this help you out?

gmartin82, 2025-05-19:
I spoke to the Cyclone team and they suggested the cause may be this linehttps://github.com/eclipse-cyclonedds/cyclonedds/blob/1e79e7d1461e7fea9a24255af1762e93ad6b0f26/src/core/ddsi/src/ddsi_receive.c#L2062

@seichmann would it be possible to get a packet capture?

gmartin82, 2025-05-19:
I usually use Wireshark to capture DDS traffic. I'm not sure of the tcpdump command but in Wireshark you can filter on `rtps` which will capture only DDS traffic.

seichmann, 2025-05-19:
[Capture-Delete.pcapng.txt](https://github.com/user-attachments/files/20298003/Capture-Delete.pcapng.txt)

seichmann, 2025-05-21:
I tried it with your branch, here is the log pointing again to the keyhash topic :)

(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x0000560048907566 in ddsi_serdata_from_keyhash (type=0x7f0ed4057b50, keyhash=0x7f0f071f98d8)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/include/dds/ddsi/ddsi_serdata.h:293
#2  0x00005600489b355a in remote_make_sample (tk=0x7f0f071f9500, gv=0x7f0ed4037730, type=0x7f0ed4057b50, vsourceinfo=0x7f0f071f95f0)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:2080
#3  0x000056004896a4bf in deliver_locally_fastpath (gv=0x7f0ed4037730, source_entity=0x7f0ec000f9b0, source_entity_locked=true, fastpath_rdary=0x7f0ec000fb20, wrinfo=0x7f0f071f9620, ops=0x56004aeeff40 <deliver_locally_ops>, 
    vsourceinfo=0x7f0f071f95f0) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_deliver_locally.c:221
#4  0x000056004896a6da in ddsi_deliver_locally_allinsync (gv=0x7f0ed4037730, source_entity=0x7f0ec000f9b0, source_entity_locked=true, fastpath_rdary=0x7f0ec000fb20, wrinfo=0x7f0f071f9620, ops=0x56004aeeff40 <deliver_locally_ops>, 
    vsourceinfo=0x7f0f071f95f0) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_deliver_locally.c:257
#5  0x00005600489b3fa9 in deliver_user_data (sampleinfo=0x7f0f24107ba0, fragchain=0x7f0f24107b18, rdguid=0x0, pwr_locked=1)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:2275
#6  0x00005600489b4076 in deliver_user_data_synchronously (sc=0x7f0f071f9d10, rdguid=0x0)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:2302
#7  0x00005600489b48df in handle_regular (rst=0x7f0f24107aa8, tnow=..., rmsg=0x7f0f24107a00, msg=0x7f0f24107a50, sampleinfo=0x7f0f071f9ea0, max_fragnum_in_msg=4294967295, rdata=0x7f0f24107b18, deferred_wakeup=0x7f0f071f9f48, 
    renew_manbypp_lease=true) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:2478
#8  0x00005600489b52cd in handle_Data (rst=0x7f0f24107aa8, tnow=..., rmsg=0x7f0f24107a00, msg=0x7f0f24107a50, size=84, sampleinfo=0x7f0f071f9ea0, keyhash=0x7f0f24107a88, datap=0x0, deferred_wakeup=0x7f0f071f9f48, 
    prev_smid=DDSI_RTPS_SMID_INFO_TS) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:2686
#9  0x00005600489b7153 in handle_submsg_sequence (thrst=0x7f0ed4036540, gv=0x7f0ed4037730, conn=0x7f0ed403a3a0, pktinfo=0x7f0f071fa1a0, tnowWC=..., tnowE=..., src_prefix=0x7f0f24107a38, dst_prefix=0x0, 
    msg=0x7f0f24107a30 "RTPS\002\005\001\001.:\001\001\374RիR\266]\342\t\001\b\000", len=116, submsg=0x7f0f24107a50 "\025\003P\000", rmsg=0x7f0f24107a00, rtps_encoded=false)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:3193
#10 0x00005600489b7923 in handle_rtps_message (thrst=0x7f0ed4036540, gv=0x7f0ed4037730, conn=0x7f0ed403a3a0, guidprefix=0x0, rbpool=0x7f0ed4043770, rmsg=0x7f0f24107a00, sz=116, 
    msg=0x7f0f24107a30 "RTPS\002\005\001\001.:\001\001\374RիR\266]\342\t\001\b\000", pktinfo=0x7f0f071fa1a0)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:3296
#11 0x00005600489b7baf in do_packet (thrst=0x7f0ed4036540, gv=0x7f0ed4037730, conn=0x7f0ed403a3a0, guidprefix=0x0, rbpool=0x7f0ed4043770)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:3394
#12 0x00005600489b8b39 in ddsi_recv_thread (vrecv_thread_arg=0x7f0ed4037ee8) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_receive.c:3679
#13 0x00005600489bc918 in create_thread_wrapper (ptr=0x7f0ed4036540) at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/core/ddsi/src/ddsi_thread.c:260
#14 0x0000560048948784 in os_startRoutineWrapper (threadContext=0x7f0ed4043930)
    at /home/XXX/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cyclors-0.3.8/cyclonedds/src/ddsrt/src/threads/posix/threads.c:198
#15 0x00007f0f24a897e2 in start_thread () from /lib64/libc.so.6
#16 0x00007f0f24b0e800 in clone3 () from /lib64/libc.so.6

gmartin82, 2025-05-21:
Thanks, with the added source line information we now know what the problem is.

gmartin82, 2025-05-21:
@seichmann Can you please pull the latest update from the branch and retest your scenario? While the plugin only handles data transfer hopefully the update fixes the crash you experienced.

seichmann, 2025-05-21:
Is there any global reason why the dispose messages are not handled/forwarded from one Zenoh Router to another inside https://github.com/eclipse-zenoh/zenoh-plugin-dds/blob/25b2f3c1a8fa5249125e5d221514bbe5b32c1f9a/zenoh-plugin-dds/src/dds_mgt.rs#L427  which only handles:

- the valid_data = true case 
- and valid_data = false + check for DDS NOT_ALIVE_DISPOSED_INSTANCE_STATE etc.

gmartin82, 2025-05-21:
Thanks for confirming the crash has been fixed.

Regarding handling of dispose messages and instance management, it's just not a feature that has been implemented yet.

PR #520 (2025-05-22): Fix segmentation fault when paired with RTI Persistence Service and Dispose on Keyed Objects
Closes #516.
closed 2025-05-22 by JEnoch